craigerl/aprsd #42: piggyback acks in message not supported

hemna, 2021-01-21:
There is a bug in the aprslib parsing library that doesn't handle parsing out the acks in line number format here:
https://github.com/rossengeorgiev/aprs-python/issues/58

This means aprsd will fail to properly parse out messages/commands when a user sends a message with {WW}AA format

```
└─[$] <> cat test-2.py
import aprslib
import pprint
msg = "KO4KWC-6>APRS,TCPIP*,qAC,T2MCI::KO4KWC-14:weather ve3kbr {22}AA"
packet = aprslib.parse(msg)
pprint.pprint(packet)
```

results in

```
└─[$] <> python test-2.py
{'addresse': 'KO4KWC-14',
 'format': 'message',
 'from': 'KO4KWC-6',
 'message_text': 'weather ve3kbr {22}AA',
 'path': ['TCPIP*', 'qAC', 'T2MCI'],
 'raw': 'KO4KWC-6>APRS,TCPIP*,qAC,T2MCI::KO4KWC-14:weather ve3kbr {22}AA',
 'to': 'APRS',
 'via': 'T2MCI'}
```

hemna, 2021-01-21:
aprsd doesn't support the piggyback ack as a result.

ldeffenb, 2022-01-12:
For additional information on where {xx}yy comes from, read: http://www.aprs.org/aprs11/replyacks.txt

hemna, 2022-01-12:
currently APRSD is not embedding acks in reply messages.

ldeffenb, 2022-01-12:
True, but it is also not acking Reply-Ack-compatible clients correctly.  Consider:

```
2022-01-12 12:50:58 EST: KJ4ERJ>APWW11,TCPIP*,qAC,SECOND::REPEAT   :n 2m{HP}
2022-01-12 12:50:58 EST: REPEAT>APZ100,TCPIP*,qAC,T2ALBERTA::KJ4ERJ   :ackHP
2022-01-12 12:50:59 EST: REPEAT>APZ100,TCPIP*,qAC,T2ALBERTA::KJ4ERJ   :K4EOC 146.8950- Toff 0.7mi W{379
2022-01-12 12:51:00 EST: KJ4ERJ>APWW11,TCPIP*,qAC,SECOND::REPEAT   :ack379

2022-01-12 12:51:32 EST: KJ4ERJ>APWW11,TCPIP*,qAC,SECOND::REPEAT   :n 2m{HP}
2022-01-12 12:51:32 EST: REPEAT>APZ100,TCPIP*,qAC,T2ALBERTA::KJ4ERJ   :ackHP
2022-01-12 12:51:32 EST: REPEAT>APZ100,TCPIP*,qAC,T2ALBERTA::KJ4ERJ   :K4EOC 146.8950- Toff 0.7mi W{380
2022-01-12 12:51:33 EST: KJ4ERJ>APWW11,TCPIP*,qAC,SECOND::REPEAT   :ack380

2022-01-12 12:52:04 EST: KJ4ERJ>APWW11,TCPIP*,qAC,SECOND::REPEAT   :n 2m{HP}
2022-01-12 12:52:04 EST: REPEAT>APZ100,TCPIP*,qAC,T2ALBERTA::KJ4ERJ   :ackHP
2022-01-12 12:52:04 EST: REPEAT>APZ100,TCPIP*,qAC,T2ALBERTA::KJ4ERJ   :K4EOC 146.8950- Toff 0.7mi W{381
2022-01-12 12:52:05 EST: KJ4ERJ>APWW11,TCPIP*,qAC,SECOND::REPEAT   :ack381
```

hemna, 2022-01-12:
Here is a current example exchange 

```
today at 12:53:43 PM Received Message _______________                                              
today at 12:53:43 PM   Raw     : KJ4ERJ>APWW11,TCPIP*,qAC,SECOND::REPEAT   :n 5 2m{HQ}
today at 12:53:43 PM    From    : KJ4ERJ                                                             
today at 12:53:43 PM    Message : n 5 2m                                                             
today at 12:53:43 PM    Msg #   : HQ                                                                 
today at 12:53:43 PM  Received Message _______________ Complete

today at 12:53:43 PM  Sending ack _______________                                                    
today at 12:53:43 PM  Raw     : REPEAT>APZ100::KJ4ERJ   :ackHQ                                     
today at 12:53:43 PM  To      : KJ4ERJ                                                             
today at 12:53:43 PM   Ack     : HQ                                                                 
today at 12:53:43 PM  Sending ack _______________ Complete

today at 12:53:44 PM  Sending Message _______________                                                 
today at 12:53:44 PM    Raw     : REPEAT>APZ100::KJ4ERJ   :K4EOC 146.8950- Toff 0.7mi W{383           
today at 12:53:44 PM    To      : KJ4ERJ                                                              
today at 12:53:44 PM    Message : K4EOC 146.8950- Toff 0.7mi W                                        
today at 12:53:44 PM    Msg #   : 383                                                                 
today at 12:53:44 PM  Sending Message _______________ Complete
```

hemna, 2024-04-12:
aprslib supports parsing out piggyback acks now.  APRSD doesn't support sending them at the moment.

hemna, 2026-08-28:
Fixed in PR #273 (`fix/issue-42-piggyback-ack-send`).

The sending half of the Reply-Ack protocol is now implemented:
- `MessagePacket._build_payload()` embeds `ackMsgNo` in the `{NNN}AAA` wire format when set
- `process_our_message_packet()` passes the incoming packet's `msgNo` as `ackMsgNo` on every plain-string reply

aprslib already parses `ackMsgNo` from inbound packets and aprsd already handles the receive side; this completes the loop.

hemna, 2026-08-28:
PR #273 has now merged (commit 1239426). Piggyback ACK (Reply-Ack) sending is live on master.

closed 2026-08-28 by hemna