craigerl/aprsd #57: KM6LYW Callsign Not showing up in GitHub search

Darrrren, 2021-04-12:
Hi Craig,

Thanks for adding the URL to the YouTube description for Patreon, I've signed up as a Patron now, hopefully you can connect the dots so you know who I am ;-)

Hey listen... Us Hams have very simple brains and can't remember anything other than 5 or 6 characters when searching for people, it's a callsign thing... So I think it might be a good idea to help everyone find you, by adding your callsign into the descriptions of your GitHub repositories.

I'm predicting a Packet Radio revival and your projects really hit the spot for us Hams who like the digital space so let's make them really easy to find and accessible whilst we're ahead of the curve.  Please add your callsign into the descriptions of everything you've done on GitHub so that they show up in the search results when people search for you.  I have no issue with the code at the moment, but an issue with the project's accessibility at this early stage... By adding your callsign to the information it should help mitigate any brand distress users may be experiencing, and make it easier for users to find you after watching your videos. Try searching for KM6LYW (your YouTube name) in Github and you'll see what I mean, 0 results! Focus on 1 search term, 1 brand, 1 call to action and promote it. More people will remember it, search for it, and more people will find you ;-)

Just an observation from a remote supporter of your project, fellow packeteer, and someone who is hopefully offering a positive contribution to help expose and promote the fantastic work that you're doing for our great hobby :-)

craigerl, 2021-04-12:
Love it - I just put my callsign in the README.md.  is that what you are thinking?  I know nothing about rising to the top of social media, so I need all the help I can get :).

shaposhnikoff, 2021-05-18:
https://grep.app/search?q=KM6LYW

closed 2024-03-12 by hemna