craigerl/aprsd #58: INSTALL.TXT improvements

Nick-DMC, 2021-04-28:
A couple adds to the INSTALL.TXT document would make it better for us beginners. 
1) Line7: virtualenv .venv_aprsd  will error our as "virtualenv command not found".  An added note to " sudo apt-get install virtualenv" would help.

2) I have an old version of Python and Line 14: pip install -e . errored out (lots of red on this one :-)  It would be helpful to mention up front that Python needs to xxx or greater.  Including how to upgrade closes the loop for us learners.

hemna, 2021-05-01:
What version of python are you using.   Aprsd only really supports python 3.6 and >

hemna, 2021-05-01:
ok I updated the INSTALL.txt to take care of the issue you mentioned and I added some instructions on how to do install the dev setup easier with the makefile.   

If you aren't a python guy or developer, then just use pypi to install it.
pip install aprsd

closed 2021-05-01 by hemna