craigerl/aprsd #78: Unable to start server

jschollenberger, 2021-12-11:
Hi!

I'm new to aprsd and just started setting it up. I pip installed, made a config, and I can't get the server started:

```Traceback (most recent call last):
  File "/usr/local/bin/aprsd", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/aprsd/aprsd.py", line 73, in main
    cli()
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/aprsd/cli_helper.py", line 65, in new_func
    return f(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/aprsd/cmds/server.py", line 70, in server
    if not client.factory.is_client_enabled():
  File "/usr/local/lib/python3.9/dist-packages/aprsd/client.py", line 196, in is_client_enabled
    enabled |= self._builders[key].is_enabled(self.config)
TypeError: unsupported operand type(s) for |=: 'bool' and 'NoneType'
```

Full output: https://gist.github.com/jschollenberger/ac8c680353f7209b91956bfec06810aa

My config is pretty vanilla. I just entered my call, aprs password, and a few other values: https://gist.github.com/jschollenberger/e850ed1be3e5b31d158a5f0c7b66db94

Thanks for any hints in the right direction.

73
Jason

hemna, 2021-12-11:
Hi Jason,
   I tested your config with the latest aprsd and I was able to reproduce the issue.   I have a fix for it here:
https://github.com/craigerl/aprsd/pull/79

If you don't want to use  aprsd from git, then you can apply a simple workaround.  
Just remove the entire kiss: block from your config for the time being and it will work.

I will release a new version of aprsd that will include the fix soon.

jschollenberger, 2021-12-11:
Hi Walter,

Thanks for the quick reply! Removing the kiss block got it going. I didn't plan on using a modem at all, just APRS-IS. Is RF a requirement?

I am currently troubleshooting why I can send messages manually, but when I receive a ping message it doesn't increment the aprsd.plugins.ping.PingPlugin processed/sent packets and the reply just spins in the "send messages" tab.

> [12/11/2021] [01:26:22 PM] [WEB_SEND_MSG-(mycall] [ERROR] XXXXXX Haven't completed everything in 60 seconds.  BAIL! - [/usr/local/lib/python3.9/dist-packages/aprsd/flask.py:256]

Jason

hemna, 2021-12-11:
Hi Jason, that typically happens when the web send message never gets an ack back from the recipient.   Do you have all the logs around your send message attempt?

hemna, 2021-12-13:
Fix is in 2.5.7 for this issue.

closed 2021-12-13 by hemna