craigerl/aprsd #80: send-message not using default config file for auth

jschollenberger, 2021-12-11:
I'm not sure if I'm using this incorrectly, but I would expect send-message to use the config file by default. The help may imply this is the intended behavior:

```
-c, --config TEXT               The aprsd config file to use for options.
                                  [default: /root/.config/aprsd/aprsd.yml]
```
```
# yq .aprs /root/.config/aprsd/aprsd.yml
{
  "enabled": true,
  "host": "rotate.aprs2.net",
  "login": mycall-15",
  "password": "12345",
  "port": 14580
}
```

but upon sending a message:

```
# aprsd send-message mycall-1 hello!
Must set --aprs_login or APRS_LOGIN
```
```
# aprsd send-message --config mycall-1 hello!
Usage: aprsd send-message [OPTIONS] TOCALLSIGN COMMAND...
Try 'aprsd send-message -h' for help.

Error: Missing argument 'COMMAND...'.
```

It does work fine when specifying the aprs login and password on the command line. Is send-message supposed to pull these values from the config? Or do they have to be provided via env variables?

hemna, 2021-12-11:
I can change it so that the fallback is to the use config if not specified.

jschollenberger, 2021-12-11:
That would be great! I'll go ahead and close this since it's not an issue but more of a feature request.

closed 2021-12-11 by jschollenberger