craigerl/aprsd #85: Cannot install on Ubuntu 18.04.6 LTS

almj, 2022-01-25:
```
`sj@radio01:~$ pip install aprsd
Collecting aprsd
  Using cached https://files.pythonhosted.org/packages/b4/d4/6dad7c183b9c8f605cc796b11535215ed6099176d9aff53b20261120d4e8/aprsd-2.5.8-py2.py3-none-any.whl
Collecting requests==2.26.0 (from aprsd)
  Using cached https://files.pythonhosted.org/packages/92/96/144f70b972a9c0eabbd4391ef93ccd49d0f2747f4f6a2a2738e99e5adc65/requests-2.26.0-py2.py3-none-any.whl
Collecting werkzeug==2.0.0 (from aprsd)
  Could not find a version that satisfies the requirement werkzeug==2.0.0 (from aprsd) (from versions: 0.1, 0.2, 0.3, 0.3.1, 0.4, 0.4.1, 0.5, 0.5.1, 0.6, 0.6.1, 0.6.2, 0.7, 0.7.1, 0.7.2, 0.8, 0.8.1, 0.8.2, 0.8.3, 0.9, 0.9.1, 0.9.2, 0.9.3, 0.9.4, 0.9.5, 0.9.6, 0.10, 0.10.1, 0.10.2, 0.10.4, 0.11, 0.11.1, 0.11.2, 0.11.3, 0.11.4, 0.11.5, 0.11.6, 0.11.7, 0.11.8, 0.11.9, 0.11.10, 0.11.11, 0.11.12, 0.11.13, 0.11.14, 0.11.15, 0.12, 0.12.1, 0.12.2, 0.13, 0.14, 0.14.1, 0.15.0, 0.15.1, 0.15.2, 0.15.3, 0.15.4, 0.15.5, 0.15.6, 0.16.0, 0.16.1, 1.0.0rc1, 1.0.0, 1.0.1)
No matching distribution found for werkzeug==2.0.0 (from aprsd)`
```

Installed `pip install Werkzeug`

```
sj@radio01:~$ pip install Werkzeug
Collecting Werkzeug
  Using cached https://files.pythonhosted.org/packages/cc/94/5f7079a0e00bd6863ef8f1da638721e9da21e5bacee597595b318f71d62e/Werkzeug-1.0.1-py2.py3-none-any.whl
Installing collected packages: Werkzeug
Successfully installed Werkzeug-1.0.1
```

But still the same issue.
What can that be?

hemna, 2022-01-25:
hrmm.   what version of python?

almj, 2022-01-25:
```
sj@radio01:~$ python --version
Python 2.7.17
sj@radio01:~$ python3 --version
Python 3.6.9
```
Got both

hemna, 2022-01-25:
try making a virtualenv with python3 and activate that and try again.   im guessing its because of py2

almj, 2022-01-25:
Sorry don't really know how, will try to google..

hemna, 2022-01-25:
do which python3.   that will show you where the binary for python executable is.

then run 
virtualenv -b /path/to/python3 .venv

source .venv/bin/activate

pip install aprsd

almj, 2022-01-25:
```
sj@radio01:~$ which python3
/usr/bin/python3
sj@radio01:~$ virtualenv -b /usr/bin/python3 .venv
Usage: virtualenv [OPTIONS] DEST_DIR

virtualenv: error: no such option: -b
```

I'm not good at this sorry...

hemna, 2022-01-26:
my bad it's -p, not -b

almj, 2022-01-26:
```
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/home/sj/.venv/lib/python3.6/site-packages/zipp.py'
Check the permissions.
```

Not my day

hemna, 2022-01-26:
most likely dude to not sourcing the virtualenv you just created

```
py3=`which python3`
virtualenv -p $py3 .venv
source .venv/bin/activate
pip install aprsd
```

hemna, 2022-01-26:
It can also be done this way.
https://linuxize.com/post/how-to-create-python-virtual-environments-on-ubuntu-18-04/

```
python3 -m venv my-project-env
source my-project-env/bin/activate
pip install aprsd
```

almj, 2022-01-26:
```
sj@radio01:~$ python3 -m venv my-project-env
sj@radio01:~$ source my-project-env/bin/activate
(my-project-env) sj@radio01:~$ pip install aprsd
Collecting aprsd
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/17/7b/1e8b838fe40aed819b8a7d29ac80b650e2b3a974884826d1e134a7aa1598/aprsd-2.5.9-py2.py3-none-any.whl (358kB)
    100% |████████████████████████████████| 368kB 4.2MB/s
Collecting cryptography==3.4.7 (from aprsd)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/9b/77/461087a514d2e8ece1c975d8216bc03f7048e6090c5166bc34115afdaa53/cryptography-3.4.7.tar.gz (546kB)
    100% |████████████████████████████████| 552kB 3.1MB/s
    Complete output from command python setup.py egg_info:

            =============================DEBUG ASSISTANCE==========================
            If you are seeing an error here please try the following to
            successfully install cryptography:

            Upgrade to the latest pip and try again. This will fix errors for most
            users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
            =============================DEBUG ASSISTANCE==========================

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-73mvawae/cryptography/setup.py", line 14, in <module>
        from setuptools_rust import RustExtension
    ModuleNotFoundError: No module named 'setuptools_rust'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-73mvawae/cryptography/
```

hemna, 2022-01-27:
try upgrading pip.

after sourcing your venv run
pip install -U pip

almj, 2022-01-28:
YES! That worked! Thank you so much for the help @hemna!

closed 2022-01-29 by hemna