craigerl/aprsd #115: NoSuchOptError when importing aprsd.threads

geowa4, 2023-05-05:
```
File ~/path/to/project/.venv/lib/python3.10/site-packages/aprsd/threads/rx.py:10
      7 from oslo_config import cfg
      9 from aprsd import client, packets, plugin
---> 10 from aprsd.threads import APRSDThread, tx
     13 CONF = cfg.CONF
     14 LOG = logging.getLogger("APRSD")

File ~/path/to/project/.venv/lib/python3.10/site-packages/aprsd/threads/tx.py:33
     29     else:
     30         _send_packet(packet, direct=direct, aprs_client=aprs_client)
---> 33 @RateLimiter(max_calls=1, period=CONF.msg_rate_limit_period, callback=limited)
     34 def _send_packet(packet: core.Packet, direct=False, aprs_client=None):
     35     if not direct:
     36         thread = SendPacketThread(packet=packet)

File ~/path/to/project/.venv/lib/python3.10/site-packages/oslo_config/cfg.py:2223, in ConfigOpts.__getattr__(self, name)
   2221     raise
   2222 except Exception:
-> 2223     raise NoSuchOptError(name)

NoSuchOptError: no such option msg_rate_limit_period in group [DEFAULT]
```

When I search through the aprsd code, I see the option defined so I'm not sure why this is happening or how to troubleshoot it further. Any help would be much appreciated.

hemna, 2023-05-05:
that's quite odd.  the default should get picked up, it's acting like it's not defined.   Do you have a config file populated?
what version of aprsd are you running and how are you launching it?

geowa4, 2023-05-05:
Thank you for the quick reply. I'll include some code snippets below to answer the questions and demonstrate my import.

```
❯ pip freeze | rg aprsd
aprsd==3.0.3
```

```
❯ rg msg_rate_limit_period ~/.config/aprsd/aprsd.conf
31:msg_rate_limit_period = 2
```

With ipython:
```
❯ ipython
Python 3.10.11 (main, May  4 2023, 19:40:24) [Clang 14.0.3 (clang-1403.0.22.14.1)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.13.2 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from aprsd_incident_response.aprsd_incident_response import IncidentResponse
---------------------------------------------------------------------------
NoSuchOptError                            Traceback (most recent call last)
```

When running the server with `aprsd server -c ~/.config/aprsd/aprsd.conf`, it seems to load though. I probably should have checked that before opening this 😬 , but it still feels weird to have this error when importing via ipython or the default Python REPL.

hemna, 2023-05-05:
I'm trying to reproduce it here and can't seem to.  I'm using python 3.10.9
I am unfamiliar with ipython but was able to install it and it didn't seem to affect the output  of running aprsd server

```
└─> aprsd server --loglevel DEBUG
05/05/2023 10:31:29 AM MainThread           INFO     APRSD is up to date                                                                                                        server.py:43
05/05/2023 10:31:29 AM MainThread           INFO     APRSD Started version: 3.0.3                                                                                               server.py:44
05/05/2023 10:31:29 AM MainThread           INFO     Loading Plugin Manager and registering plugins                                                                             server.py:54
05/05/2023 10:31:29 AM MainThread           INFO     Loading APRSD Plugins                                                                                                     plugin.py:454
05/05/2023 10:31:29 AM MainThread           INFO     Email services not enabled.                                                                                                 email.py:98
05/05/2023 10:31:29 AM MainThread           WARNING  Plugin EmailPlugin is disabled                                                                                            plugin.py:430
05/05/2023 10:31:29 AM MainThread           INFO     Registering Regex plugin 'aprsd.plugins.fortune.FortunePlugin'(3.0.3) -- ^([f]|[f]\s|fortune)                             plugin.py:421
05/05/2023 10:31:29 AM MainThread           INFO     Registering Regex plugin 'aprsd.plugins.location.LocationPlugin'(3.0.3) -- ^([l]|[l]\s|location)                          plugin.py:421
05/05/2023 10:31:29 AM MainThread           INFO     Registering Regex plugin 'aprsd.plugins.ping.PingPlugin'(3.0.3) -- ^([p]|[p]\s|ping)                                      plugin.py:421
05/05/2023 10:31:29 AM MainThread           INFO     Registering Regex plugin 'aprsd.plugins.query.QueryPlugin'(3.0.3) -- ^\!.*                                                plugin.py:421
05/05/2023 10:31:29 AM MainThread           INFO     Registering Regex plugin 'aprsd.plugins.time.TimePlugin'(3.0.3) -- ^([t]|[t]\s|time)                                      plugin.py:421
05/05/2023 10:31:29 AM MainThread           INFO     Registering Regex plugin 'aprsd.plugins.weather.OWMWeatherPlugin'(3.0.3) -- ^([w]|[w]\s|[w][x]|[w][x]\s|weather)          plugin.py:421
05/05/2023 10:31:29 AM MainThread           INFO     Registering Regex plugin 'aprsd.plugins.version.VersionPlugin'(3.0.3) -- ^([v]|[v]\s|version)                             plugin.py:421
05/05/2023 10:31:29 AM MainThread           DEBUG    ==> setup_connection: call "{'self': <aprsd.client.APRSISClient object at 0x110249c30>}"                                    trace.py:48
05/05/2023 10:31:29 AM MainThread           INFO     Creating aprslib client                                                                                                   client.py:145
05/05/2023 10:31:29 AM MainThread           INFO     Attempting connection to noam.aprs2.net:14580                                                                               inet.py:226
05/05/2023 10:31:30 AM MainThread           INFO     Connected to ('44.10.10.10', 14580)                                                                                         inet.py:233
05/05/2023 10:31:30 AM MainThread           DEBUG    Banner: # aprsc 2.1.14-g5e22b37                                                                                             inet.py:246
05/05/2023 10:31:30 AM MainThread           INFO     Sending login information                                                                                                 aprsis.py:109
05/05/2023 10:31:30 AM MainThread           DEBUG    Server: # logresp WB4BOR-12 verified, server T2SJC                                                                        aprsis.py:119
05/05/2023 10:31:30 AM MainThread           INFO     Connected to T2SJC                                                                                                        aprsis.py:128
05/05/2023 10:31:30 AM MainThread           INFO     Login successful                                                                                                          aprsis.py:142
05/05/2023 10:31:30 AM MainThread           DEBUG    Logging in to APRS-IS with user 'WB4BOR-12'                                                                               client.py:161
05/05/2023 10:31:30 AM MainThread           DEBUG    <== setup_connection: return (719ms) <aprsd.clients.aprsis.Aprsdis object at 0x110249db0>
```

hemna, 2023-05-05:
I can reproduce it if I run ipython and then simply try and import aprsd.threads.rx it will happen.  That being said, 
that is an invalid way to get the threads.rx as it's assumed that rx is being used by one of the aprsd commands in 
aprsd.cmds

If I run ipython then 

```
└─> ipython
Python 3.10.9 (main, Dec 12 2022, 18:49:42) [Clang 14.0.0 (clang-1400.0.29.202)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.13.2 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from aprsd.threads import rx
---------------------------------------------------------------------------
NoSuchOptError                            Traceback (most recent call last)
File ~/devel/mine/hamradio/aprsd/.venv/lib/python3.10/site-packages/oslo_config/cfg.py:2219, in ConfigOpts.__getattr__(self, name)
   2218 try:
-> 2219     return self._get(name)
   2220 except ValueError:

File ~/devel/mine/hamradio/aprsd/.venv/lib/python3.10/site-packages/oslo_config/cfg.py:2653, in ConfigOpts._get(self, name, group, namespace)
   2652         pass
-> 2653 value, loc = self._do_get(name, group, namespace)
   2654 self.__cache[key] = value

File ~/devel/mine/hamradio/aprsd/.venv/lib/python3.10/site-packages/oslo_config/cfg.py:2671, in ConfigOpts._do_get(self, name, group, namespace)
   2669     return (self.GroupAttr(self, self._get_group(name)), None)
-> 2671 info = self._get_opt_info(name, group)
   2672 opt = info['opt']

File ~/devel/mine/hamradio/aprsd/.venv/lib/python3.10/site-packages/oslo_config/cfg.py:2876, in ConfigOpts._get_opt_info(self, opt_name, group)
   2875 if not real_opt_name:
-> 2876     raise NoSuchOptError(opt_name, group)
   2877 log_real_group_name = real_group_name or 'DEFAULT'

NoSuchOptError: no such option msg_rate_limit_period in group [DEFAULT]
```

If you are trying to import this directly from an ipython prompt, you have to import the aprsd conf first.

```
└─> ipython
Python 3.10.9 (main, Dec 12 2022, 18:49:42) [Clang 14.0.0 (clang-1400.0.29.202)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.13.2 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from aprsd import conf

In [2]: from aprsd.threads import rx

In [3]:
```

hemna, 2023-05-05:
I just commited a change in tx.py to import the aprsd.conf to ensure that the conf entry is defined.
install from the latest against git and see if that fixes the problem for you.

geowa4, 2023-05-12:
Works well. Thank you so much!

closed 2023-05-12 by geowa4