craigerl/aprsd #117: Twitter Messaging Plugin Issue

CtrlAltDel-Irl, 2023-06-02:
It seems the virtual radio isn't picking up the messages destined for twitter.  I haven't tried with another plugin for full disclosure.  Using aprsd version 3.0.3 and python 3.9.2

I tried sending 2 messages formatted like :

t  Testing twitter aprsd plugin

sent to MI7DJT-12  (my virtual radio in aprsd)

As you can see in the logfile there is no activity picked up by the virtual radio.

I have attached logfile and aprsd.yml

hemna, 2023-07-08:
I don't see the attachment for the log

CtrlAltDel-Irl, 2023-07-09:
Hey Walter,

I don't have this equipment now so can't run logs.  I did send them months
ago though on github.

hemna, 2023-07-09:
I just fired up the latest version of aprsd and enabled the twitter plugin and aprsd was able to get the request over the aprs network to send a tweet.  The tweet command when the twitter plugin is enabled and configured is 'tw <tweet message>'.  

That works, but the problem is the twitter API has been altered to need a paying subscription.

You can see my log here:
[image]

When aprsd starts up, it tries to load all of the plugins as defined in the config file.   When a plugin loads correctly you should see something like this.

```
07/09/2023 09:20:50 AM MainThread           INFO     Registering Regex plugin 'aprsd_twitter_plugin.twitter.SendTweetPlugin'(0.5.0) -- plugin.py:421
                                                     ^([t][w]\s|twitter)
```

When a plugin fails to load you would see this

```
07/09/2023 09:20:49 AM MainThread           ERROR    Failed to load Plugin 'aprsd_slack_plugin.notify_plugin' : 'No module named       plugin.py:376
```