craigerl/aprsd #121: Web Interface Not Starting

UsedCows, 2023-07-16:
Running aprsd v3.0.4 with the config file properly configured (to the best of my knowledge) to enable the web interface. Yet when launching, no joy. I have even tried changing the web_ip value to the actual IP address of the device. NMAP shows port 8001 down regardless. When launching, I get the string of responses (i.e. plugins enabled, etc.) but no mention of the web interface and whether it was or was not enabled.

Am I perhaps missing something else in the config file necessary to enable?

```
web_enabled = true
web_ip = 0.0.0.0
web_port = 8001
user = admin
password = password
```

Thanks in advance!
~Christian
N1COW

hemna, 2023-07-16:
Hi Christian, 
You are correct, it isn't started automatically.  Unfortunately, for stability reasons aprsd no longer starts up as part of the server command.   It's a flask based service and it was having issues shutting down properly with multiple threads running. 

So I separated it out to make the server command more stable.   You can still get the web admin UI running using gunicorn.
I also need to update the documentation and provide a sample gunicorn setup to run the web interface.  Give me a few minutes and I can provide you the info on how to start it properly

hemna, 2023-07-16:
ok I just put in a change in master that allows. you to easily start the web admin interface

First create a virtual environment, source the venv, install aprsd, install gunicorn, then launch the web interface

```
virtualenv .venv
source .venv/bin/activate
cd <aprsd git repo>
pip install .
pip install gunicorn
gunicorn -b :8080 "aprsd.wsgi:app"
```

This starts the web interface on port 8080.

UsedCows, 2023-07-23:
I have tried your suggestions. However, I seem to have run into an issue. When starting gunicorn, it fails. Among the console output is the following. 

```
    raise ConfigFilesNotFoundError(self._namespace._files_not_found)
oslo_config.cfg.ConfigFilesNotFoundError: Failed to find some config files: /config/aprsd.conf
```
Should I have an aprsd.conf file? Ido have the config.yml, of course. But an aprsd.conf file I do not. 

Advice?

hemna, 2023-07-23:
I am reworking the code for running the web admin interface.   I should have something done before long.

hemna, 2023-07-24:
Ok can you give this a try?

Update your local git repo and re-install aprsd in your local virtualenvironment.

then source your virtual env and run

python -m aprsd.wsgi

UsedCows, 2023-07-24:
Sadly, still experiencing the same issue

```
(.venv_aprsd) n8tgc@NUC:~/aprsd2/aprsd$ python -m aprsd.wsgi
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/n8tgc/aprsd2/aprsd/aprsd/wsgi.py", line 329, in <module>
    log_level = init_app(log_level="DEBUG")
  File "/home/n8tgc/aprsd2/aprsd/aprsd/wsgi.py", line 314, in init_app
    CONF(
  File "/home/n8tgc/.venv_aprsd/lib/python3.8/site-packages/oslo_config/cfg.py", line 2156, in __call__
    raise ConfigFilesNotFoundError(self._namespace._files_not_found)
oslo_config.cfg.ConfigFilesNotFoundError: Failed to find some config files: /home/n8tgc/.config/aprsd/aprsd.conf
```
Failed to find aprsd.conf

hemna, 2023-07-25:
> (.venv_aprsd) ***@***.***:~/aprsd2/aprsd$ python -m aprsd.wsgi

you need to create your aprsd.conf file

run aprsd sample-config > aprsd.conf

then edit your file and put it in $HOME/.config/aprsd/aprsd.conf

UsedCows, 2023-07-26:
I was able to test a little this morning. A few observations...

1. The console output appears to be in DEBUG mode. 
2. It looks like the sequence starts and immediately restarts (restarting with stat)

```
07/26/2023 05:41:33 AM MainThread      DEBUG    email_plugin.smtp_password     = ****                                                                      cfg.py:2609
07/26/2023 05:41:33 AM MainThread      DEBUG    email_plugin.smtp_port         = 465                                                                       cfg.py:2609
07/26/2023 05:41:33 AM MainThread      DEBUG    email_plugin.smtp_use_ssl      = True                                                                      cfg.py:2609
07/26/2023 05:41:33 AM MainThread      DEBUG    ********************************************************************************                           cfg.py:2613
 * Serving Flask app 'aprsd'
 * Debug mode: on
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on http://127.0.0.1:8001
Press CTRL+C to quit
 * Restarting with stat
07/26/2023 05:41:34 AM MainThread      DEBUG    ********************************************************************************                           cfg.py:2589
07/26/2023 05:41:34 AM MainThread      DEBUG    Configuration options gathered from:                                                                       cfg.py:2590
07/26/2023 05:41:34 AM MainThread      DEBUG    command line args: []                                                                                      cfg.py:2591
07/26/2023 05:41:34 AM MainThread      DEBUG    config files: ['/home/n8tgc/.config/aprsd/aprsd.conf']                                                     cfg.py:2592
07/26/2023 05:41:34 AM MainThread      DEBUG    ================================================================================                           cfg.py:2594
```
3.The web-interface is functional but only from 127.0.0.1. Attempting to access it from the device's network IP address fails from another network device. NMAP from another device shows port 8001 to be closed. All of this despite declaring the IP address in the aprsd.conf file. Or, should I NOT un-comment these lines?

```
web_enabled = true
web_ip = 10.79.68.100
web_port = 8001
```
4. Regardless of the performance of the web interface, it appears no connectivity is made to aprs.fi. In all previous testing, I was able to perfform a ping/pong test and get a fortune response from another aprs device. Today, when I send a packet from a radio to this test aprsd server, I don't even get confirmation on the console output that the packet was received from aprs.fi

```
enabled = true
login = N1COW-4
password = 10652
host = noam.aprs2.net
port = 14580
```

Hope these bits of information help

hemna, 2023-07-26:
I have disabled debug mode.  That will fix the double starting.  I have also updated the startup so that it uses the web_ip defined in the config for the listen host/port.

UsedCows, 2023-07-26:
Making progress!

I have aprsd up and running and able to access the web interface. :)

It appears that I am still not able to connect to an APRS-FI server, When starting and monitoring the data displayed on the console, I do not see an attempt to connect to an APRS-FI server. Nor do I see a failure to connect. See image below. Also, uptime is undefined. Not a big deal, just an observation. I also have aprsd v2.6.1 running on a nearby RPi i use for comparison. 

[image]

```
#From aprsd.conf
enabled = true
login = N1COW-4
password = 10652
host = noam.aprs2.net
port = 14580

apiKey = (correct key)
```

hemna, 2023-07-26:
how are you starting the server?

UsedCows, 2023-07-26:
(.venv_aprsd) n1cow@LinuxLaptop:~/.venv_aprsd/bin$ python -m aprsd.wsgi

hemna, 2023-07-26:
> (.venv_aprsd) ***@***.***:~/.venv_aprsd/bin$ python -m aprsd.wsgi

ok thats just the web interface for watching things that happen when
running the listen or server commands.

run

aprsd —help

aprsd server —loglevel DEBUG

hemna, 2023-07-26:
hey no problemo.  I'm glad I could help :)