craigerl/aprsd #126: Installation failure

UgoPoddine, 2023-07-26:
Hello,

I was using aprsd without special issue on a RaspberryPI4 sine yesterday, installed using pip3
I was forced to rebuild the image, but now the installation failed with the following error.
I tried also : 
a) to install criptography from repository
b) to install rust separately
c) to install openssl from repository

... but the same error is still there.

System is : 
6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

Python : 
Python 3.9.2

pip3 : 
pip 23.2.1 

Can you help ?
Thank-you
73 IU1IPB ugo

      running build_ext
      running build_rust

          =============================DEBUG ASSISTANCE=============================
          If you are seeing a compilation error please try the following steps to
          successfully install cryptography:
          1) Upgrade to the latest pip and try again. This will fix errors for most
             users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
          2) Read https://cryptography.io/en/latest/installation/ for specific
             instructions for your platform.
          3) Check our frequently asked questions for more information:
             https://cryptography.io/en/latest/faq/
          4) Ensure you have a recent Rust toolchain installed:
             https://cryptography.io/en/latest/installation/#rust

          Python: 3.9.2
          platform: Linux-6.1.21+-armv6l-with-glibc2.31
          pip: n/a
          setuptools: 68.0.0
          setuptools_rust: 1.6.0
          rustc: n/a
          =============================DEBUG ASSISTANCE=============================

      error: can't find Rust compiler

      If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.

      To update pip, run:

          pip install --upgrade pip

      and then retry package installation.

      If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.

      This package requires Rust >=1.56.0.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects

hemna, 2023-07-26:
yah this is not surprising.  The problem is with the cryptography library being built with rustc instead of C now and unfortunately, most debian based systems have rustc and the build chain for rustc apps is very very old and doesn't work.   

https://forums.raspberrypi.com/viewtopic.php?t=343187

hemna, 2023-07-26:
This link suggests https://stackoverflow.com/questions/66118337/how-to-get-rid-of-cryptography-build-error

```
# upgrade pip
pip install --upgrade pip

# install rust toolchain
curl https://sh.rustup.rs -sSf | sh -s -- --default-toolchain stable -y
PATH="/root/.cargo/bin:${PATH}"

# build from source
pip install cryptography --no-cache-dir --only-binary=:all: --force-reinstall --upgrade
```

UgoPoddine, 2023-07-27:
Hello, 

Thanks.
As I wrote,  pip is at its last version.
I was a bit worried to install the full toolchain on a weak Raspberry and to build from source, but I will try and I will let you know. 

Ugo

hemna, 2023-08-27:
I have removed the need for cryptography now that webchat doesn't do ssl by default.  You should be able to install this just fine now.

hemna, 2023-09-17:
I am closing this due to lack of feedback from the original reporter as well as the removal of the cryptography dependency that caused the issue.

closed 2023-09-17 by hemna