craigerl/aprsd #127: Weather Plugin - Incorrect Location

UsedCows, 2023-07-28:
Now running v3.1.1 (thank you again). Things going great. However, the weather plugin appears to not be providing the correct location of sender. 

Previously on 1.6.0 & 2.6.1 simply sending weather to the aprsd station would respond as expected w/local weather. On 3.1.1, it appears that the position provided is somwehere near Antwerp. (I'm in New Hampshire) I tried multiple plugins. Seems like the problem is consistent. Unless, of course, I am missing something obvious. 

Am I?

```
#USMetarPlugin
07/28/2023 07:38:18 AM ProcessPKT           INFO     WX Plugin 'Metar'                                                                                  weather.py:108
07/28/2023 07:38:19 AM ProcessPKT           ERROR    Couldn't fetch forecast.weather.gov '404 Client Error: Not Found for url:                          weather.py:146
                                                     https://api.weather.gov/points/51.0767,4.4892'                                                                   
#USWeatherPlugin
07/28/2023 07:49:29 AM ProcessPKT           INFO     OWMWeather Plugin 'Weather'                                                                        weather.py:210
07/28/2023 07:49:29 AM ProcessPKT           ERROR    Couldn't fetch openweathermap api '401 Client Error: Unauthorized for url:                         weather.py:245
                                                     https://api.openweathermap.org/data/2.5/onecall?lat=51.07667&lon=4.48917&appid=27fb3342fcd4c021ee1               
                                                     60b7a69d96baf&units=imperial&exclude=minutely,hourly'     
```

UsedCows, 2023-07-31:
I do have the OWM apiKey already provided and un-commented. Also, the plugin appears to be correctly loaded.
 
```
07/31/2023 05:34:59 PM MainThread           INFO     Registering Regex plugin 'aprsd.plugins.weather.OWMWeatherPlugin'(3.1.1) -- ^([w][x]|[w][x]\s|weather)                                                                     plugin.py:421
...
07/31/2023 05:34:59 PM MainThread           INFO     <aprsd.plugins.weather.OWMWeatherPlugin object at 0x7ffa44a3ea30>                                                                                                           server.py:6
```
However, when sending 'weather', the coordinates provided are near Anterp

```
07/31/2023 05:35:27 PM ProcessPKT           INFO     OWMWeatherPlugin called                                                                                                                                                    plugin.py:210
07/31/2023 05:35:27 PM ProcessPKT           INFO     OWMWeather Plugin 'Weather'                                                                                                                                               weather.py:210
07/31/2023 05:35:28 PM ProcessPKT           ERROR    Couldn't fetch openweathermap api '401 Client Error: Unauthorized for url:                                                                                                weather.py:245
                                                     https://api.openweathermap.org/data/2.5/onecall?lat=51.07667&lon=4.48917&appid={correct API key}&units=imperial&exclude=minutely,hourly'
```
Sending 'weather n1cow-9' does pass along the correct location coordinates, but the results are the same from OWM. Looks like there are two issues at play. One is OWM responding with a 401 error, regardless of the location provided. The second issue is the Anterp coordinates appearing when not including a station. If a position beacon was not running on the callsign running aprsd could that explain the odd location? I assumed that not sending a station with the 'weather' message, it would default to the location of the sending station. Perhaps, aprsd is defaulting to the location of the callsign associated with aprsd?

```
07/31/2023 05:40:09 PM ProcessPKT           INFO     OWMWeather Plugin 'Weather n1cow-9'                                                                                                                                       weather.py:210
07/31/2023 05:40:10 PM ProcessPKT           ERROR    Couldn't fetch openweathermap api '401 Client Error: Unauthorized for url:                                                                                                weather.py:245
                                                     https://api.openweathermap.org/data/2.5/onecall?lat=42.82233&lon=-71.62783&appid={correct API key}&units=imperial&exclude=minutely,hourly'
```
FWIW, I was able to confirm that the API key is working by trying the sample API call from the OWM site.

`https://api.openweathermap.org/data/2.5/weather?lat=42.82233&lon=-71.62783&appid={correct API key}&exclude=minutely,hourly`

The only difference I can see is '/2.5/onecall?' in aprsd and '/2.5/weather?' in the sample from OWM.

hemna, 2023-08-01:
I just got home and tested the OWMWeatherPlugin from the latest master code.

I used the dev test-plugin command

```
╭─aprsd on  master [$?] via 🐍 v3.10.9 (.aprsd-venv) took 2s ❯
└─> aprsd dev test-plugin --loglevel DEBUG -p aprsd.plugins.weather.OWMWeatherPlugin weather n1cow-9
...
...
...
07/31/2023 08:58:36 PM MainThread           INFO     OWMWeather Plugin 'weather n1cow-9'                                              weather.py:210
07/31/2023 08:58:36 PM MainThread           DEBUG    Fetch aprs.fi location for 'N1COW-9'                                         plugin_utils.py:12
07/31/2023 08:58:37 PM MainThread           DEBUG    Fetch openweathermap for 42.82233, -71.62783                                 plugin_utils.py:65
07/31/2023 08:58:37 PM MainThread           DEBUG    <== process: return (499ms) 'few clouds 65.2F/57.5F Wind 5@308G12 76%'              trace.py:80
07/31/2023 08:58:37 PM MainThread           INFO     Result0 = '['few clouds 65.2F/57.5F Wind 5@308G12 76%']'                             dev.py:131
```

hemna, 2023-08-01:
I just tried generating a new api key and used both onecall and weather urls and they both failed with 401.  OWM must have crippled their API ?

hemna, 2023-08-01:
I can't even find information on the 2.5 API.   I am thinking that OpenWeather isn't so Open now.  Looks like you have to pay for the one call interface now.   If this is the case, I might just remove the OpenWeatherMap plugin entirely from APRSD and make it a separate plugin.

For what it's worth I personally use the AVWX Plugin and run my own instance of avwx as a container for my aprsd instance, so I can have free weather reports anywhere in the world.

https://avwx.rest/

https://github.com/craigerl/aprsd/blob/master/aprsd/plugins/weather.py#L281

hemna, 2023-08-01:
ok so FWIW, I have re-enabled the old school MapClick.php api in the USWeatherPlugin to see if it works.  I have tested it here locally and it seems to work.   Update from master and try that

UsedCows, 2023-08-01:
Confirmed that USWeather is now working. (yay!)

Regarding AVWXWeather, I have a question. When configured, I am still getting a response that the plugin is not enabled. 

```
enabled_plugins = aprsd.plugins.weather.AVWXWeatherPlugin

...

8/01/2023 08:27:16 AM MainThread           INFO     Loading APRSD Plugins                                                                               plugin.py:454
08/01/2023 08:27:16 AM MainThread           WARNING  Plugin AVWXWeatherPlugin is disabled                                                                plugin.py:430
08/01/2023 08:27:16 AM MainThread           INFO     Completed Plugin Loading.                                                                           plugin.py:470
```

Out of curiosity, I tried 'enabling' it

```
[avwx_plugin]
enabled = true
```

Yeah. That didn't work. 

```
08/01/2023 08:32:19 AM MainThread           INFO     AVWXWeatherPlugin called                                                                            plugin.py:210
08/01/2023 08:32:19 AM MainThread           WARNING  AVWXWeatherPlugin isn't enabled                                                                     plugin.py:213
```

How to enable?

Thank you again for the USWeather fix.

hemna, 2023-08-03:
Did you specify the base_url and the apikey?

closed 2023-08-11 by hemna