craigerl/aprsd #141: No config file found

alexvotofuture, 2023-11-28:
Using release 3.2.1 (installed with Pip per install instructions), when I try to run `aprsd sample-config > ~/.aprs/config.yml` or `aprsd sample-config > ~/.config/aprsd/config.yml` the sample config file is generated and copied to the folder, but when I attempt to run `aprsd server` I keep receiving an error stating:
> "No config file found!! run aprsd sample-config"
and the server fails to initialize.

On release 3.2.2 (also installed via Pip) the sample config file will not copy at all, and I receive an error beginning with "WARNING".

How do I correctly copy the sample-config file to a folder where it can be found and get the server to locate it?

Please forgive me if I've made an obvious error - I'm not a coder by profession, just a HAM and hobbyist who's played with RasPi and Arduinos a bit before. I've been trying for months to build something like this, and just came upon your solution, so I'm excited to get this working to play with it. Thanks!

hemna, 2023-11-28:
First off,
  running aprsd sample-config dumps the config to standard out, so you can see the config.   You can simply redirect that to a file.

aprsd sample-config > aprsd.conf
aprsd server -c ./aprsd.conf

The standard place where aprsd looks for it's config is in ~/.config/aprsd/aprsd.conf

So you can simply do this:

mkdir ~/.config/aprsd
aprsd sample-config > ~/.config/aprsd/aprsd.conf
aprsd server

alexvotofuture, 2023-11-29:
Thank you so much for the detailed instructions, @hemna! Followed them and finally got the server up and running. These could definitely be useful in the readme/docs. Closing the issue.

closed 2023-11-29 by alexvotofuture