craigerl/aprsd #142: Email plugin infinite re-send

alexvotofuture, 2023-12-18:
When using the email plugin, as soon as there's an email available to be read from my inbox via IMAP, the APRSD server keeps re-sending the same email message over APRS-IS multiple times. (See photo below)

I started getting errors like "Failed to UnPickle /home/.../packettrack.p" and "'MessagePacket' object has no attribute 'msgNo'. I noticed there was no packettrack.p file auto-created in my .config/aprsd file. When I tried creating an empty packettrack.p file, it started populating with packet information after the first re-send, but I got the same errors as before and the re-send issue persisted.

With or without the packettrack.p file, I receive an error before the first re-send only, saying that in the email.py file there's a "FutureWarning: Possible nested set at position [3 and 9] f= re.search("

Any idea whether I should manually create the packettrack.p file again, how it should be set up (is there an example somewhere?), or whether there's some issue with that f=re.search command that's causing received emails messages to endlessly re-send?

Running APRSD version 3.2.1

[image]

hemna, 2023-12-19:
can you paste the exception/log with the MessagePacket has no msgNo error?

hemna, 2023-12-19:
You can simply delete the packettrack.p file.   It's just for saving some state between aprsd starts and stops.

alexvotofuture, 2023-12-19:
> can you paste the exception/log with the MessagePacket has no msgNo error?
Appears when there is an empty packettrack.p file available:
[image]

With or without the packettrack.p file the server immediately starts re-sending the message in the inbox when started, even though it's sent it many times before.

Side note: Is there a quick way to disable auto-sending/re-sending email, so that it just waits for the command to be sent via APRS to deliver new messages?

Thanks again for being so responsive on here!

alexvotofuture, 2023-12-19:
Another interesting thing: When I deleted that message from my inbox, the next time I started up the server I started receiving a flood of ack messages from APRS.fi for all the previous re-send attempts.

hemna, 2023-12-19:
ok I'm able to reproduce the ERROR    Failed to UnPickle /Users/I530566/.config/aprsd//packettrack.p by simply removing all the contents of the file, but leaving it in place.   

One thing that you can do is the next time you start aprsd server pass the -f
-f will flush all the old data files that it creates and it will start clean.  It won't delete your config though.

When you restart aprsd, does it still want to keep sending the same email?

alexvotofuture, 2023-12-19:
Nice, glad you were able to reproduce the error.

I passed the -f flag and restarted the server. Same issue as before. Getting that nesting error, and then a bunch of repeat tx's.
[image]

hemna, 2023-12-19:
I cant really read your log, since it's an image.   when I click on it, i get a 404 error.

alexvotofuture, 2023-12-19:
Created the log file and enabled logging. Here's another round in text:
[12/18/2023 06:47:52 PM] [MainThread          ] [WARNI] Version 3.2.1 of aprsd is outdated. Version 3.2.2 was released Wednesday November 22, 2023. - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:42]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] APRSD Started version: 3.2.1 - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:45]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Loading Plugin Manager and registering plugins - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:55]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Loading APRSD Plugins - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:454]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Email shortcuts {REDACTED} - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugins/email.py:96]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Registering Regex plugin 'aprsd.plugins.email.EmailPlugin'(3.2.1) -- ^-.* - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:421]
[12/18/2023 06:47:52 PM] [MainThread          ] [WARNI] Plugin FortunePlugin is disabled - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:430]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Registering Regex plugin 'aprsd.plugins.location.LocationPlugin'(3.2.1) -- ^([l]|[l]\s|location) - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:421]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Registering Regex plugin 'aprsd.plugins.ping.PingPlugin'(3.2.1) -- ^([p]|[p]\s|ping) - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:421]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Registering Regex plugin 'aprsd.plugins.query.QueryPlugin'(3.2.1) -- ^\!.* - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:421]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Registering Regex plugin 'aprsd.plugins.time.TimePlugin'(3.2.1) -- ^([t]|[t]\s|time) - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:421]
[12/18/2023 06:47:52 PM] [MainThread          ] [ERROR] Config.owm_weather_plugin.apiKey is not set.  Disabling - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugins/weather.py:198]
[12/18/2023 06:47:52 PM] [MainThread          ] [WARNI] Plugin OWMWeatherPlugin is disabled - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:430]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Registering Regex plugin 'aprsd.plugins.version.VersionPlugin'(3.2.1) -- ^([v]|[v]\s|version) - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:421]
[12/18/2023 06:47:52 PM] [MainThread          ] [WARNI] Plugin NotifySeenPlugin is disabled - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:418]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Completed Plugin Loading. - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/plugin.py:470]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Message Plugins enabled and running: - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:63]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] <aprsd.plugins.time.TimePlugin object at 0x72acd448> - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:65]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] <aprsd.plugins.version.VersionPlugin object at 0x7270f8c8> - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:65]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] <aprsd.plugins.ping.PingPlugin object at 0x72b52af0> - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:65]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] <aprsd.plugin.HelpPlugin object at 0x7270f118> - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:65]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] <aprsd.plugins.location.LocationPlugin object at 0x72b52940> - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:65]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] <aprsd.plugins.email.EmailPlugin object at 0x72bff160> - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:65]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] <aprsd.plugins.query.QueryPlugin object at 0x72acd3a0> - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:65]
[12/18/2023 06:47:52 PM] [MainThread          ] [INFO ] Watchlist Plugins enabled and running: - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/cmds/server.py:66]
[12/18/2023 06:47:52 PM] [KeepAlive           ] [INFO ] KN6UYO-10 - Uptime 00:00:00 RX:0 TX:0 Tracker:0 Msgs TX:0 RX:0 Last:00:00:00 Email: N/A - RAM Current:4KB Peak:4KB Threads:2 - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/threads/keep_alive.py:66]
[12/18/2023 06:47:52 PM] [RX_MSG              ] [INFO ] Creating APRS client - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/client.py:53]
[... cut ...]
[12/18/2023 06:57:45 PM] [MainThread          ] [INFO ] WatchList::Saving 1 entries to disk at/home/alessandrovoto/.config/aprsd - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/utils/objectstore.py:71]
[12/18/2023 06:57:45 PM] [MainThread          ] [INFO ] SeenList::Saving 1 entries to disk at/home/alessandrovoto/.config/aprsd - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/utils/objectstore.py:71]
[12/18/2023 06:57:45 PM] [MainThread          ] [INFO ] Uptime:0:09:53.334549 Msgs TX:6 RX:0 ACK: TX:0 RX:0 Email TX:0 RX:0 LastLoop:2023-12-18 18:57:43.203186  - [/home/alessandrovoto/.local/lib/python3.9/site-packages/aprsd/main.py:94]

alexvotofuture, 2023-12-19:
Another random thought - could this have something to do with the packet_dupe_timeout in the config file? Mine is set to 30 currently.

hemna, 2023-12-19:
any chance you can rerun this with debug logging?

aprsd server --loglevel DEBUG

hemna, 2024-03-12:
Any updates on this issue for you?

alexvotofuture, 2024-07-15:
Hey @hemna, I'm so sorry I never got back to you on this after you jumped in to help me out.

Ultimately I gave up on the email plugin and just used the custom plugin template you included in the package, then modified it to talk to Twilio's API to pass messages received through APRS to any cell number. I haven't figured out two-way communication, but at least this got me to a proof of concept where I could get a message out from remote areas in case of emergency.

Thanks again for your help on this particular issue, and for making an awesome and flexible tool I can use and learn from!

closed 2024-07-15 by alexvotofuture