craigerl/aprsd #161: Fresh pip install crashes on `sample-config`

petrus-lt, 2024-04-21:
aprsd was installed as user or root, either with `pip install aprsd` or `pip install .` from cloned repo. I'm having the same issue under both Linux/arm64 and FreeBSD/amd64.

```
petrus@rpi4:~ $ aprsd sample-config
Traceback (most recent call last):
File "/home/petrus/.local/bin/aprsd", line 8, in <module>
sys.exit(main())
File "/home/petrus/.local/lib/python3.9/site-packages/aprsd/main.py", line 81, in main
cli(auto_envvar_prefix="APRSD")
File "/home/petrus/.local/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
return self.main(*args, **kwargs)
File "/home/petrus/.local/lib/python3.9/site-packages/click/core.py", line 1078, in main
rv = self.invoke(ctx)
File "/home/petrus/.local/lib/python3.9/site-packages/click/core.py", line 1688, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/home/petrus/.local/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "/home/petrus/.local/lib/python3.9/site-packages/click/core.py", line 783, in invoke
return __callback(*args, **kwargs)
File "/home/petrus/.local/lib/python3.9/site-packages/click/decorators.py", line 33, in new_func
return f(get_current_context(), *args, **kwargs)
File "/home/petrus/.local/lib/python3.9/site-packages/aprsd/main.py", line 136, in sample_config
args = get_namespaces()
File "/home/petrus/.local/lib/python3.9/site-packages/aprsd/main.py", line 128, in get_namespaces
selected = imp.entry_points(group="oslo.config.opts")
TypeError: entry_points() got an unexpected keyword argument 'group'
```

Moreover, running `aprsd server` ignores the config i've put in `.config/aprsd/aprsd.yml`.  
Is there a missing dependency, or something else?

hemna, 2024-04-21:
Looks like python 3.9, is your cloned repo of aprsd up to date?  git pull ?

petrus-lt, 2024-04-21:
Yup py3.9, and yes i'm up to date:

```
~/tools/aprsd (master*) » git rev-parse HEAD
c43652dbea3c55243b1457ce3abba235d50fcea1
```

hemna, 2024-04-21:
I just released 3.3.4 with a fix.  try upgrading and give that a shot.

pip install -U aprsd

petrus-lt, 2024-04-21:
Thanks, it now works! Didn't realize 3.9 was old... I don't python much.

closed 2024-04-21 by hemna