craigerl/aprsd #165: module 'aprsd.packets.core' has no attribute 'ThirdParty'

slyticoon, 2024-05-20:
Constant attribute errors when parsing packets from IP TNC Direwolf:

RX_PKT             | ERROR    | Failed to parse bytes received from KISS interface. | aprsd.clients.kiss:parse_frame:81
2024-05-19 22:21:44.007 | RX_PKT             | ERROR    | module 'aprsd.packets.core' has no attribute 'ThirdParty' | aprsd.clients.kiss:parse_frame:82
Traceback (most recent call last):

  File "/usr/lib/python3.9/threading.py", line 912, in _bootstrap
    self._bootstrap_inner()
    │    └ <function Thread._bootstrap_inner at 0xf75c4730>
    └ <APRSDPluginRXThread(RX_PKT, started 4065326144)>
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
    │    └ <function APRSDThread.run at 0xf65174a8>
    └ <APRSDPluginRXThread(RX_PKT, started 4065326144)>
  File "/home/jesse/.local/lib/python3.9/site-packages/aprsd/threads/aprsd.py", line 51, in run
    can_loop = self.loop()
               │    └ <function APRSDRXThread.loop at 0xf63a12f8>
               └ <APRSDPluginRXThread(RX_PKT, started 4065326144)>
  File "/home/jesse/.local/lib/python3.9/site-packages/aprsd/threads/rx.py", line 46, in loop
    self._client.consumer(
    │    │       └ <function KISSClient.consumer at 0xf63ff0b8>
    │    └ <aprsd.client.KISSClient object at 0xf38b4d48>
    └ <APRSDPluginRXThread(RX_PKT, started 4065326144)>
  File "/home/jesse/.local/lib/python3.9/site-packages/aprsd/client.py", line 356, in consumer
    self._client.consumer(callback)
    │    │       │        └ <bound method APRSDRXThread._process_packet of <APRSDPluginRXThread(RX_PKT, started 4065326144)>>
    │    │       └ <function KISS3Client.consumer at 0xf63fb5c8>
    │    └ <aprsd.clients.kiss.KISS3Client object at 0xf38b4208>
    └ <aprsd.client.KISSClient object at 0xf38b4d48>
  File "/home/jesse/.local/lib/python3.9/site-packages/aprsd/clients/kiss.py", line 87, in consumer
    self.kiss.read(callback=self.parse_frame, min_frames=None)
    │    │    │             │    └ <function KISS3Client.parse_frame at 0xf63fb580>
    │    │    │             └ <aprsd.clients.kiss.KISS3Client object at 0xf38b4208>
    │    │    └ <function SyncFrameDecode.read at 0xf644d100>
    │    └ TCPKISS(_protocol=KISSProtocol(transport=<_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>, frames=<Queue ...
    └ <aprsd.clients.kiss.KISS3Client object at 0xf38b4208>
  File "/home/jesse/.local/lib/python3.9/site-packages/ax253/decode.py", line 248, in read
    self.protocol.read_frames(
    │    └ <property object at 0xf6445cf8>
    └ TCPKISS(_protocol=KISSProtocol(transport=<_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>, frames=<Queue ...
  File "/home/jesse/.local/lib/python3.9/site-packages/ax253/decode.py", line 143, in read_frames
    return loop.run_until_complete(_())
           │    │                  └ <function FrameDecodeProtocol.read_frames.<locals>._ at 0xf2e9e970>
           │    └ <function BaseEventLoop.run_until_complete at 0xf69e4850>
           └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/lib/python3.9/asyncio/base_events.py", line 629, in run_until_complete
    self.run_forever()
    │    └ <function BaseEventLoop.run_forever at 0xf69e4808>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/lib/python3.9/asyncio/base_events.py", line 596, in run_forever
    self._run_once()
    │    └ <function BaseEventLoop._run_once at 0xf69e55c8>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1890, in _run_once
    handle._run()
    │      └ <function Handle._run at 0xf6a3bb68>
    └ <Handle <TaskWakeupMethWrapper object at 0xf38b47f0>(<Future finished result=None>)>
  File "/usr/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
    │    │            │    │           │    └ <member '_args' of 'Handle' objects>
    │    │            │    │           └ <Handle <TaskWakeupMethWrapper object at 0xf38b47f0>(<Future finished result=None>)>
    │    │            │    └ <member '_callback' of 'Handle' objects>
    │    │            └ <Handle <TaskWakeupMethWrapper object at 0xf38b47f0>(<Future finished result=None>)>
    │    └ <member '_context' of 'Handle' objects>
    └ <Handle <TaskWakeupMethWrapper object at 0xf38b47f0>(<Future finished result=None>)>
  File "/home/jesse/.local/lib/python3.9/site-packages/ax253/decode.py", line 141, in _
    return [f async for f in self.read(n_frames=n_frames, callback=callback)]
                             │    │             │                  └ <bound method KISS3Client.parse_frame of <aprsd.clients.kiss.KISS3Client object at 0xf38b4208>>
                             │    │             └ None
                             │    └ <function FrameDecodeProtocol.read at 0xf643da48>
                             └ KISSProtocol(transport=<_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>, frames=<Queue at 0xf2e91d90 maxs...
  File "/home/jesse/.local/lib/python3.9/site-packages/ax253/decode.py", line 141, in <listcomp>
    return [f async for f in self.read(n_frames=n_frames, callback=callback)]
            │           └ b'\x82\xa0\x9e\xa8\xaa`\xe0\xaeh\x86\x82\xa8@\xe0\xaeb\x82\xa4\x9c@\xe2\xae\x92\x88\x8ab@\xe0\xae\x92\x88\x8ad@c\x03\xf0!3608...
            └ b'\x82\xa0\x9e\xa8\xaa`\xe0\xaeh\x86\x82\xa8@\xe0\xaeb\x82\xa4\x9c@\xe2\xae\x92\x88\x8ab@\xe0\xae\x92\x88\x8ad@c\x03\xf0!3608...
  File "/home/jesse/.local/lib/python3.9/site-packages/ax253/decode.py", line 123, in read
    callback(frame)
    │        └ b'\x82\xa0\xa8\xa8h@`\xaeh\xa8\x8c\xae@b\x9cp\x88\x8a\xaa@\xee\xae\x92\x88\x8ab@\xe0\xaeh\x8e\x8e\x9a@\xe2\xaeb\x82\xa4\x9c@\...
    └ <bound method KISS3Client.parse_frame of <aprsd.clients.kiss.KISS3Client object at 0xf38b4208>>
> File "/home/jesse/.local/lib/python3.9/site-packages/aprsd/clients/kiss.py", line 79, in parse_frame
    self._parse_callback(**kwargs)
    │    │                 └ {'frame': Frame(destination=Address(callsign=b'APTT4', ssid=0, digi=False), source=Address(callsign=b'W4TFW', ssid=1, digi=Fa...
    │    └ <bound method APRSDRXThread._process_packet of <APRSDPluginRXThread(RX_PKT, started 4065326144)>>
    └ <aprsd.clients.kiss.KISS3Client object at 0xf38b4208>
  File "/home/jesse/.local/lib/python3.9/site-packages/aprsd/threads/rx.py", line 70, in _process_packet
    self.process_packet(*args, **kwargs)
    │    │               │       └ {'frame': Frame(destination=Address(callsign=b'APTT4', ssid=0, digi=False), source=Address(callsign=b'W4TFW', ssid=1, digi=Fa...
    │    │               └ ()
    │    └ <function APRSDDupeRXThread.process_packet at 0xf63c6898>
    └ <APRSDPluginRXThread(RX_PKT, started 4065326144)>
  File "/home/jesse/.local/lib/python3.9/site-packages/aprsd/threads/rx.py", line 97, in process_packet
    packet = self._client.decode_packet(*args, **kwargs)
             │    │       │              │       └ {'frame': Frame(destination=Address(callsign=b'APTT4', ssid=0, digi=False), source=Address(callsign=b'W4TFW', ssid=1, digi=Fa...
             │    │       │              └ ()
             │    │       └ <function KISSClient.decode_packet at 0xf63ff028>
             │    └ <aprsd.client.KISSClient object at 0xf38b4d48>
             └ <APRSDPluginRXThread(RX_PKT, started 4065326144)>
  File "/home/jesse/.local/lib/python3.9/site-packages/aprsd/client.py", line 345, in decode_packet
    if isinstance(packet, core.ThirdParty):
                  │       └ <module 'aprsd.packets.core' from '/home/jesse/.local/lib/python3.9/site-packages/aprsd/packets/core.py'>
                  └ WeatherPacket(_type='WeatherPacket', from_call='W4TFW-1', to_call='APTT4', addresse=None, format='uncompressed', msgNo=None, ...

AttributeError: module 'aprsd.packets.core' has no attribute 'ThirdParty'

samlachance, 2024-05-21:
I am also experiencing this using Direwolf over IP

samlachance, 2024-06-03:
I believe the bug was introduced with this commit and stems from some incorrect object management: https://github.com/craigerl/aprsd/commit/ae9e4d31ad8333451d6e81dd6440085052c192a0

@slyticoon I would create a commit to fix this but I'm not a python guy and am not familiar enough with this project to confidently change things without regressions. My suggestion to get it working immediately would be to clone the repo, create a branch for this temporary fix, revert that entire commit, and use as expected.

p4ck377r4c3r, 2024-08-09:
Yes as packed this is throwing AttributeError when installed via pip. I have quieted the error by changing line 28 in core.py to PACKET_TYPE_THIRDPARTY = "ThirdParty". Unsure if this is the actual fix but it does get rid of the errors. still testing this program as I just found it yesterday and think it will be a great asset to my digi.

hemna, 2024-09-16:
hey guys, I just pushed a patch to master to fix this.  thanks for the heads up.

ffa28fa28adc60375ec0ca03f730fc321f09bfc6

hemna, 2024-09-30:
Fix released in 3.4.1

closed 2024-09-30 by hemna