craigerl/aprsd #172: I'm a little bit lost

9M2PJU, 2024-08-05:
9m2pju@raspi-200:~/.config/aprsd $ aprsd server --config /home/9m2pju/.config/aprsd/aprsd.conf
2024-08-05 17:11:33.831 | MainThread         | INFO     | APRSD is up to date | aprsd.cmds.server:server:47
2024-08-05 17:11:33.846 | MainThread         | INFO     | APRSD Started version: 3.4.0 | aprsd.cmds.server:server:48
2024-08-05 17:11:33.866 | MainThread         | INFO     | Creating client connection | aprsd.cmds.server:server:58
2024-08-05 17:11:33.888 | MainThread         | INFO     | Creating aprslib client(rotate.aprs2.net:14580) and logging in 9M2PJU. | aprsd.client:setup_connection:238
2024-08-05 17:11:33.903 | MainThread         | INFO     | Attempting connection to rotate.aprs2.net:14580 | aprslib.inet:_connect:226
2024-08-05 17:11:34.134 | MainThread         | INFO     | Connected to ('2607:f130:0:154::6d58:458e', 14580, 0, 0) | aprslib.inet:_connect:233
2024-08-05 17:11:34.553 | MainThread         | INFO     | Login successful | aprsd.clients.aprsis:_send_login:144
2024-08-05 17:11:34.567 | MainThread         | INFO     | Connected to T2PANAMA | aprsd.clients.aprsis:_send_login:146
2024-08-05 17:11:34.581 | MainThread         | INFO     | <aprsd.client.APRSISClient object at 0xb4246c70> | aprsd.cmds.server:server:60
2024-08-05 17:11:34.594 | MainThread         | INFO     | Loading Plugin Manager and registering plugins | aprsd.cmds.server:server:66
2024-08-05 17:11:34.615 | MainThread         | INFO     | Loading APRSD Plugins | aprsd.plugin:setup_plugins:479
2024-08-05 17:11:34.641 | MainThread         | INFO     | Email services not enabled. | aprsd.plugins.email:setup:130
2024-08-05 17:11:34.655 | MainThread         | WARNING  | Plugin EmailPlugin is disabled | aprsd.plugin:_load_plugin:455
2024-08-05 17:11:34.671 | MainThread         | INFO     | Fortune path None | aprsd.plugins.fortune:setup:25
2024-08-05 17:11:34.685 | MainThread         | WARNING  | Plugin FortunePlugin is disabled | aprsd.plugin:_load_plugin:455
2024-08-05 17:11:34.702 | MainThread         | INFO     | Registering Regex plugin 'aprsd.plugins.location.LocationPlugin'(3.4.0) -- ^([l]|[l]\s|location) | aprsd.plugin:_load_plugin:446
2024-08-05 17:11:34.726 | MainThread         | INFO     | Registering Regex plugin 'aprsd.plugins.ping.PingPlugin'(3.4.0) -- ^([p]|[p]\s|ping) | aprsd.plugin:_load_plugin:446
2024-08-05 17:11:34.751 | MainThread         | ERROR    | Failed to load Plugin 'aprsd.plugins.query' : 'No module named 'aprsd.plugins.query'' | aprsd.plugin:_create_class:401
2024-08-05 17:11:34.767 | MainThread         | INFO     | Registering Regex plugin 'aprsd.plugins.time.TimePlugin'(3.4.0) -- ^([t]|[t]\s|time) | aprsd.plugin:_load_plugin:446
2024-08-05 17:11:34.792 | MainThread         | ERROR    | Config.owm_weather_plugin.apiKey is not set.  Disabling | aprsd.plugins.weather:setup:201
2024-08-05 17:11:34.808 | MainThread         | WARNING  | Plugin OWMWeatherPlugin is disabled | aprsd.plugin:_load_plugin:455
2024-08-05 17:11:34.824 | MainThread         | INFO     | Registering Regex plugin 'aprsd.plugins.version.VersionPlugin'(3.4.0) -- ^([v]|[v]\s|version) | aprsd.plugin:_load_plugin:446
2024-08-05 17:11:34.853 | MainThread         | WARNING  | Plugin NotifySeenPlugin is disabled | aprsd.plugin:_load_plugin:443
2024-08-05 17:11:34.869 | MainThread         | INFO     | Completed Plugin Loading. | aprsd.plugin:setup_plugins:495
2024-08-05 17:11:34.996 | MainThread         | INFO     | Message Plugins enabled and running: | aprsd.cmds.server:server:74
2024-08-05 17:11:35.010 | MainThread         | INFO     | <aprsd.plugins.version.VersionPlugin object at 0xb413ba48> | aprsd.cmds.server:server:76
2024-08-05 17:11:35.023 | MainThread         | INFO     | <aprsd.plugin.HelpPlugin object at 0xb413b898> | aprsd.cmds.server:server:76
2024-08-05 17:11:35.036 | MainThread         | INFO     | <aprsd.plugins.location.LocationPlugin object at 0xb413b8f8> | aprsd.cmds.server:server:76
2024-08-05 17:11:35.050 | MainThread         | INFO     | <aprsd.plugins.ping.PingPlugin object at 0xb413b958> | aprsd.cmds.server:server:76
2024-08-05 17:11:35.063 | MainThread         | INFO     | <aprsd.plugins.time.TimePlugin object at 0xb413b9e8> | aprsd.cmds.server:server:76
2024-08-05 17:11:35.076 | MainThread         | INFO     | Watchlist Plugins enabled and running: | aprsd.cmds.server:server:77
2024-08-05 17:11:35.125 | MainThread         | INFO     | Beacon Enabled.  Starting Beacon thread. | aprsd.cmds.server:server:130
2024-08-05 17:11:35.159 | MainThread         | INFO     | Beacon thread is running and will send beacons every 1800 seconds. | aprsd.threads.tx:__init__:228
2024-08-05 17:11:35.195 | MainThread         | INFO     | Registry Enabled.  Starting Registry thread. | aprsd.cmds.server:server:135
2024-08-05 17:11:35.229 | MainThread         | INFO     | APRS Registry thread is running and will send info every 3600 seconds to https://aprs.hemna.com/api/v1/registry. | aprsd.threads.registry:__init__:30
Sending log entries 2
2024-08-05 17:11:35.429 | LogMonitorThread   | WARNING  | Failed to send log entries 2 | aprsd.threads.log_monitor:send_log_entries:31
2024-08-05 17:11:35.460 | LogMonitorThread   | WARNING  | HTTPConnectionPool(host='0.0.0.0', port=8001): Max retries exceeded with url: /log_entries (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb413f5e0>: Failed to establish a new connection: [Errno 111] Connection refused')) | aprsd.threads.log_monitor:send_log_entries:32
Sending log entries 35
2024-08-05 17:11:37.700 | LogMonitorThread   | WARNING  | Failed to send log entries 35 | aprsd.threads.log_monitor:send_log_entries:31
2024-08-05 17:11:37.729 | LogMonitorThread   | WARNING  | HTTPConnectionPool(host='0.0.0.0', port=8001): Max retries exceeded with url: /log_entries (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb413f760>: Failed to establish a new connection: [Errno 111] Connection refused')) | aprsd.threads.log_monitor:send_log_entries:32
Sending log entries 2
2024-08-05 17:11:39.894 | LogMonitorThread   | WARNING  | Failed to send log entries 2 | aprsd.threads.log_monitor:send_log_entries:31
2024-08-05 17:11:39.925 | LogMonitorThread   | WARNING  | HTTPConnectionPool(host='0.0.0.0', port=8001): Max retries exceeded with url: /log_entries (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb413b8b0>: Failed to establish a new connection: [Errno 111] Connection refused')) | aprsd.threads.log_monitor:send_log_entries:32
Sending log entries 2
2024-08-05 17:11:42.075 | LogMonitorThread   | WARNING  | Failed to send log entries 2 | aprsd.threads.log_monitor:send_log_entries:31
2024-08-05 17:11:42.105 | LogMonitorThread   | WARNING  | HTTPConnectionPool(host='0.0.0.0', port=8001): Max retries exceeded with url: /log_entries (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb413fb50>: Failed to establish a new connection: [Errno 111] Connection refused')) | aprsd.threads.log_monitor:send_log_entries:32
2024-08-05 17:11:43.179 | StatsStore         | INFO     | StatsStore::Saving 9 entries to disk at /home/9m2pju/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
Sending log entries 3
2024-08-05 17:11:45.332 | LogMonitorThread   | WARNING  | Failed to send log entries 3 | aprsd.threads.log_monitor:send_log_entries:31
2024-08-05 17:11:45.361 | LogMonitorThread   | WARNING  | HTTPConnectionPool(host='0.0.0.0', port=8001): Max retries exceeded with url: /log_entries (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb413f520>: Failed to establish a new connection: [Errno 111] Connection refused')) | aprsd.threads.log_monitor:send_log_entries:32
Sending log entries 2
2024-08-05 17:11:47.513 | LogMonitorThread   | WARNING  | Failed to send log entries 2 | aprsd.threads.log_monitor:send_log_entries:31
2024-08-05 17:11:47.543 | LogMonitorThread   | WARNING  | HTTPConnectionPool(host='0.0.0.0', port=8001): Max retries exceeded with url: /log_entries (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb413ff40>: Failed to establish a new connection: [Errno 111] Connection refused')) | aprsd.threads.log_monitor:send_log_entries:32
Sending log entries 2
2024-08-05 17:11:49.726 | LogMonitorThread   | WARNING  | Failed to send log entries 2 | aprsd.threads.log_monitor:send_log_entries:31
[... cut ...]
2024-08-05 17:11:59.778 | LogMonitorThread   | WARNING  | Failed to send log entries 2 | aprsd.threads.log_monitor:send_log_entries:31
2024-08-05 17:11:59.807 | LogMonitorThread   | WARNING  | HTTPConnectionPool(host='0.0.0.0', port=8001): Max retries exceeded with url: /log_entries (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb413bf70>: Failed to establish a new connection: [Errno 111] Connection refused')) | aprsd.threads.log_monitor:send_log_entries:32

is there any wrong i have done?

hemna, 2024-08-05:
As a workaround you can disable the web admin ui in the config file by setting...

```
[admin]

#
# From aprsd.conf
#

# Enable the Admin Web Interface (boolean value)
web_enabled = false
```

9M2PJU, 2024-08-05:
Thanks a lot!

I am trying to make a telegram bot using aprsd-telegram-plugin.

hemna, 2024-08-05:
ok I haven't used that plugin in a while.  If you find any issues let me know and I can try to address them.

9M2PJU, 2024-08-05:
what is the correct format for lat, long? i have set 344.2344444. Do it need to change to aprx format?

9M2PJU, 2024-08-05:
2024-08-05 22:45:33.323 | MainThread         | ERROR    | Failed to load Plugin 'aprsd_telegram_plugin.telegram' : 'cannot import name 'objectstore' from 'aprsd' (/home/9m2pju/.local/lib/python3.9/site-packages/aprsd/__init__.py)' | aprsd.plugin:_create_class:401

i think this plugin needs to be updated by the developer

9M2PJU, 2024-08-05:
9m2pju@raspi-200:~/aprsd $ aprsd server
load config
2024-08-05 23:14:30.448 | MainThread         | INFO     | APRSD is up to date | aprsd.cmds.server:server:48
2024-08-05 23:14:30.466 | MainThread         | INFO     | APRSD Started version: 3.4.1.dev16+g4b26e2b.d20240805 | aprsd.cmds.server:server:49
2024-08-05 23:14:30.485 | MainThread         | INFO     | Creating client connection | aprsd.cmds.server:server:58
2024-08-05 23:14:30.504 | MainThread         | INFO     | Creating aprslib client(rotate.aprs2.net:14580) and logging in 9M2PJU. | aprsd.client.aprsis:setup_connection:104
2024-08-05 23:14:30.517 | MainThread         | INFO     | Attempting connection to rotate.aprs2.net:14580 | aprslib.inet:_connect:226
2024-08-05 23:14:30.759 | MainThread         | INFO     | Connected to ('2607:f130:0:154::6d58:458e', 14580, 0, 0) | aprslib.inet:_connect:233
2024-08-05 23:14:31.164 | MainThread         | INFO     | Login successful | aprsd.client.drivers.aprsis:_send_login:144
2024-08-05 23:14:31.176 | MainThread         | INFO     | Connected to T2PANAMA | aprsd.client.drivers.aprsis:_send_login:146
2024-08-05 23:14:31.188 | MainThread         | INFO     | <aprsd.client.aprsis.APRSISClient object at 0xb5f65808> | aprsd.cmds.server:server:60
2024-08-05 23:14:31.200 | MainThread         | INFO     | Loading Plugin Manager and registering plugins | aprsd.cmds.server:server:66
2024-08-05 23:14:31.219 | MainThread         | INFO     | Loading APRSD Plugins | aprsd.plugin:setup_plugins:479
2024-08-05 23:14:31.243 | MainThread         | INFO     | Email services not enabled. | aprsd.plugins.email:setup:130
2024-08-05 23:14:31.256 | MainThread         | WARNING  | Plugin EmailPlugin is disabled | aprsd.plugin:_load_plugin:455
2024-08-05 23:14:31.269 | MainThread         | INFO     | Fortune path None | aprsd.plugins.fortune:setup:25
2024-08-05 23:14:31.282 | MainThread         | WARNING  | Plugin FortunePlugin is disabled | aprsd.plugin:_load_plugin:455
2024-08-05 23:14:31.297 | MainThread         | INFO     | Registering Regex plugin 'aprsd.plugins.location.LocationPlugin'(3.4.1.dev16+g4b26e2b.d20240805) -- ^([l]|[l]\s|location) | aprsd.plugin:_load_plugin:446
2024-08-05 23:14:31.319 | MainThread         | INFO     | Registering Regex plugin 'aprsd.plugins.ping.PingPlugin'(3.4.1.dev16+g4b26e2b.d20240805) -- ^([p]|[p]\s|ping) | aprsd.plugin:_load_plugin:446
2024-08-05 23:14:31.344 | MainThread         | ERROR    | Failed to load Plugin 'aprsd.plugins.query' : 'No module named 'aprsd.plugins.query'' | aprsd.plugin:_create_class:401
2024-08-05 23:14:31.358 | MainThread         | INFO     | Registering Regex plugin 'aprsd.plugins.time.TimePlugin'(3.4.1.dev16+g4b26e2b.d20240805) -- ^([t]|[t]\s|time) | aprsd.plugin:_load_plugin:446
2024-08-05 23:14:31.380 | MainThread         | ERROR    | Config.owm_weather_plugin.apiKey is not set.  Disabling | aprsd.plugins.weather:setup:201
2024-08-05 23:14:31.393 | MainThread         | WARNING  | Plugin OWMWeatherPlugin is disabled | aprsd.plugin:_load_plugin:455
2024-08-05 23:14:31.406 | MainThread         | INFO     | Registering Regex plugin 'aprsd.plugins.version.VersionPlugin'(3.4.1.dev16+g4b26e2b.d20240805) -- ^([v]|[v]\s|version) | aprsd.plugin:_load_plugin:446
2024-08-05 23:14:31.429 | MainThread         | WARNING  | Plugin NotifySeenPlugin is disabled | aprsd.plugin:_load_plugin:443
2024-08-05 23:14:31.441 | MainThread         | INFO     | Completed Plugin Loading. | aprsd.plugin:setup_plugins:495
2024-08-05 23:14:31.560 | MainThread         | INFO     | Message Plugins enabled and running: | aprsd.cmds.server:server:74
2024-08-05 23:14:31.572 | MainThread         | INFO     | <aprsd.plugins.time.TimePlugin object at 0xb4129e68> | aprsd.cmds.server:server:76
2024-08-05 23:14:31.584 | MainThread         | INFO     | <aprsd.plugins.location.LocationPlugin object at 0xb4129cb8> | aprsd.cmds.server:server:76
2024-08-05 23:14:31.596 | MainThread         | INFO     | <aprsd.plugin.HelpPlugin object at 0xb4129ad8> | aprsd.cmds.server:server:76
2024-08-05 23:14:31.608 | MainThread         | INFO     | <aprsd.plugins.ping.PingPlugin object at 0xb4129dc0> | aprsd.cmds.server:server:76
2024-08-05 23:14:31.619 | MainThread         | INFO     | <aprsd.plugins.version.VersionPlugin object at 0xb41295f8> | aprsd.cmds.server:server:76
2024-08-05 23:14:31.631 | MainThread         | INFO     | Watchlist Plugins enabled and running: | aprsd.cmds.server:server:77
2024-08-05 23:14:31.676 | MainThread         | INFO     | Beacon Enabled.  Starting Beacon thread. | aprsd.cmds.server:server:126
2024-08-05 23:14:31.708 | MainThread         | INFO     | Beacon thread is running and will send beacons every 1800 seconds. | aprsd.threads.tx:__init__:228
2024-08-05 23:14:31.740 | MainThread         | INFO     | Registry Enabled.  Starting Registry thread. | aprsd.cmds.server:server:131
2024-08-05 23:14:31.772 | MainThread         | INFO     | APRS Registry thread is running and will send info every 3600 seconds to https://aprs.hemna.com/api/v1/registry. | aprsd.threads.registry:__init__:30
2024-08-05 23:14:39.727 | StatsStore         | INFO     | StatsStore::Saving 9 entries to disk at /home/9m2pju/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
2024-08-05 23:14:49.801 | StatsStore         | INFO     | StatsStore::Saving 9 entries to disk at /home/9m2pju/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
[... cut ...]
2024-08-05 23:16:40.605 | StatsStore         | INFO     | StatsStore::Saving 9 entries to disk at /home/9m2pju/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
2024-08-05 23:16:50.674 | StatsStore         | INFO     | StatsStore::Saving 9 entries to disk at /home/9m2pju/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
2024-08-05 23:17:00.743 | StatsStore         | INFO     | StatsStore::Saving 9 entries to disk at /home/9m2pju/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78

9M2PJU-1 still not appear on aprs.fi

hemna, 2024-08-06:
It looks like the entire API for telegram has changed since I wrote the initial plugin.   I'm working on reworking the code to work with the new API.

9M2PJU, 2024-08-06:
Thanks OM.

hemna, 2024-08-06:
This library looks promising

https://github.com/eternnoir/pyTelegramBotAPI

But this will require a complete rewrite of the plugin to work.

9M2PJU, 2024-08-20:
Thanks OM.