craigerl/aprsd #173: Tcp KISS

p4ck377r4c3r, 2024-08-10:
When aprsd only has tcp kiss enabled, it  gives attribute error. I was able to fix this by changing Line 92 in kiss.py to all lowercase. Once the attribute error was fixed, it looks asit does when running on the 'IS' connection. However, it doesn't seem to be parsing the "thirdparty" packets and never receives a ping message over kiss. I would look further into fixing this and contribute but not knowing the work flow is my limiting factor. Would take me a bit, but would figure it out. Everything seems to work fine when  connection IS enabled and tcp disabled. 

I am running direwolf that seems to send everything over the kiss port as third party. I have tested APRSISCE with the same connection and it seems to work fine.

hemna, 2024-08-13:
hey there, thanks for the info.   The TCPKISS connection is via 3rd party library that aprsd uses.  

Here is the encapsulation being done by aprsd to send the packet to ax253 lib.

https://github.com/craigerl/aprsd/blob/master/aprsd/client/drivers/kiss.py#L90-L119

hemna, 2024-10-29:
Does this issue still persist for you?

If so, can you provide the attribute error from the logs please?

hemna, 2025-04-24:
Master branch has an entirely new KISS TCP Client.  I wrote it from scratch to help with a new feature I'm adding.  If you are still using this, please let me know and let me know if the problem persists.  I'll close this issue if I don't hear back.