craigerl/aprsd #175: OWM not working

9M2PJU, 2024-08-22:
added apiKey but unable to get weather on aprs messaging.

hemna, 2024-08-22:
Can you please provide debug logs?

bizkut, 2024-08-22:
```
2024-08-22 15:07:26.357 | RX_PKT             | INFO     | RX <- MessagePacket:76 9M2PJU-5 ->TCPIP*->qAC->9M2PJU-> 9M2PJU-2 : W | aprsd.packets.log:log:142
2024-08-22 15:07:26.374 | ProcessPKT         | DEBUG    | ProcessPKT-LOOP 1906 | aprsd.threads.rx:process_packet:183
2024-08-22 15:07:26.390 | TXAck-9M2PJU-5-76  | DEBUG    | Starting | aprsd.threads.aprsd:run:48
2024-08-22 15:07:26.397 | TXAck-9M2PJU-5-76  | INFO     | TX -> AckPacket:76 (1 of 3) 9M2PJU-2 -> 9M2PJU-5 | aprsd.packets.log:log:142
2024-08-22 15:07:26.401 | ProcessPKT         | DEBUG    | VersionPlugin called | aprsd.plugin:filter:211
2024-08-22 15:07:26.407 | ProcessPKT         | DEBUG    | OWMWeatherPlugin called | aprsd.plugin:filter:211
2024-08-22 15:07:26.412 | ProcessPKT         | INFO     | OWMWeather Plugin 'W' | aprsd.plugins.weather:process:220
2024-08-22 15:07:26.418 | ProcessPKT         | DEBUG    | Fetch aprs.fi location for '9M2PJU-5' | aprsd.plugin_utils:get_aprs_fi:12
2024-08-22 15:07:26.690 | KeepAlive          | INFO     | 9M2PJU-2 - Uptime 0:31:52.605920 RX:18 TX:36 Tracker:36 Msgs TX:0 RX:0 Last:00:00:00 Email: N/A - RAM Current:711KB Peak:810KB Threads:7 | aprsd.threads.keep_alive:loop:77
2024-08-22 15:07:26.693 | KeepAlive          | INFO     | KeepAlive       Alive? True  0:00:00.003665       | aprsd.threads.keep_alive:loop:87
2024-08-22 15:07:26.697 | KeepAlive          | INFO     | StatsStore      Alive? True  0:00:00.136583       | aprsd.threads.keep_alive:loop:87
2024-08-22 15:07:26.703 | KeepAlive          | INFO     | RX_PKT          Alive? True  0:00:00.562913       | aprsd.threads.keep_alive:loop:87
2024-08-22 15:07:26.706 | KeepAlive          | INFO     | ProcessPKT      Alive? True  0:00:00.518066       | aprsd.threads.keep_alive:loop:87
2024-08-22 15:07:26.708 | KeepAlive          | INFO     | BeaconSendThread Alive? True  0:00:00.883868       | aprsd.threads.keep_alive:loop:87
2024-08-22 15:07:26.711 | KeepAlive          | INFO     | APRSRegistryThread Alive? True  0:00:00.812492       | aprsd.threads.keep_alive:loop:87
2024-08-22 15:07:26.714 | KeepAlive          | INFO     | TXAck-9M2PJU-5-76 Alive? True  0:00:00.300586       | aprsd.threads.keep_alive:loop:87
2024-08-22 15:07:27.293 | ProcessPKT         | DEBUG    | Fetch openweathermap for 3.17350, 101.72883 | aprsd.plugin_utils:fetch_openweathermap:65
2024-08-22 15:07:27.301 | ProcessPKT         | DEBUG    | Fetching OWM weather 'https://api.openweathermap.org/data/2.5/onecall?lat=3.17350&lon=101.72883&appid=5084077518b14e4c45765ca97fae18d3&units=imperial&exclude=minutely,hourly' | aprsd.plugin_utils:fetch_openweathermap:79
2024-08-22 15:07:27.562 | ProcessPKT         | ERROR    | Couldn't fetch openweathermap api '401 Client Error: Unauthorized for url: https://api.openweathermap.org/data/2.5/onecall?lat=3.17350&lon=101.72883&appid=5084077518b14e4c45765ca97fae18d3&units=imperial&exclude=minutely,hourly' | aprsd.plugins.weather:process:255
2024-08-22 15:07:27.569 | ProcessPKT         | DEBUG    | TimePlugin called | aprsd.plugin:filter:211
2024-08-22 15:07:27.572 | ProcessPKT         | DEBUG    | PingPlugin called | aprsd.plugin:filter:211
2024-08-22 15:07:27.575 | ProcessPKT         | DEBUG    | LocationPlugin called | aprsd.plugin:filter:211
2024-08-22 15:07:27.578 | ProcessPKT         | DEBUG    | YahooStockQuote called | aprsd.plugin:filter:211
2024-08-22 15:07:27.584 | ProcessPKT         | DEBUG    | HelpPlugin called | aprsd.plugin:filter:211
2024-08-22 15:07:27.593 | ProcessPKT         | DEBUG    | Sending 'Unable to get weather' | aprsd.threads.rx:process_our_message_packet:318
2024-08-22 15:07:27.600 | TX-9M2PJU-5-530    | DEBUG    | Starting | aprsd.threads.aprsd:run:48
2024-08-22 15:07:27.604 | TX-9M2PJU-5-530    | INFO     | TX -> MessagePacket:530 (1 of 3) 9M2PJU-2 -> 9M2PJU-5 : Unable to get weather | aprsd.packets.log:log:142
```

9M2PJU, 2024-08-22:
Can you please rewrite and change to free MET Noway API OM ?

https://api.met.no/

hemna, 2024-09-01:
Looks like something is wrong with your auth key

```'401 Client Error: Unauthorized for url:```

closed 2024-10-03 by hemna