craigerl/aprsd #176: Question about RF query

9M2PJU, 2024-08-23:
Hello OM,
If I add callsign = *, will aprsd listen to any callsign from RF ?

hemna, 2024-09-01:
I'm not sure what you are referring to?   Is this a setting in the config file?

9M2PJU, 2024-10-01:
yes OM

hemna, 2024-10-01:
You have to set callsign to the callsign you want APRSD to send packets as.
If you want to listen to any packets coming from a tcpkiss connection, then
run the aprsd listen command, after you have configured aprsd.conf to point to the tcpkiss server.

9M2PJU, 2024-10-01:
So i need to have a tcpkiss server to listen to any callsign who send query?

hemna, 2024-10-01:
You mentioned aprsd listening to RF.  There are 2 ways to do that, via a TCPKISS connection or a SerialKISS connection.

The other options is if aprsd has internet access, then it can connect to an aprs server over a tcp socket which is in the 
aprs_network section of the config file.   It's up to you to decide which is best for you and what you are trying to accomplish.

You need a valid legal ham callsign to use aprsd to connect to aprs as it will transmit acks for packets recieved for the configured callsign.

9M2PJU, 2024-10-02:
Thanks OM. But if i'm using the internet, will the aprsd reply to any
callsign who sends a query to it or just callsign that i have listed on the
config only?

hemna, 2024-10-02:
If there is an igate within listening distance of your radio, yes APRSD will get the message and respond to it accordingly

Like I said before, the callsign configured in your config is the callsign of the APRSD instance.

9M2PJU, 2024-10-02:
Ok, thanks for the clarification. but how about this line

# The Ham callsign to allow access to the query plugin from RF.
# (string value)
callsign =

means any listed callsign in the (callsign = X0XXX) should accept a
response from aprsd ?

hemna, 2024-10-03:
No.  It's exactly what the comment says. 
```
# The Ham callsign to allow access to the query plugin from RF.
```
The query plugin allowed the admin/owner of the APRSD instance to take actions on the running aprsd instance.

closed 2024-10-03 by hemna