craigerl/aprsd #190: Full python path to the class name list

9M2PJU, 2025-03-15:
Hello OM, 
Can you add all the class names for the external plugins on the README ?

hemna, 2025-04-15:
I can add the path in the README for each of the external plugins.  would that be ok?

hemna, 2025-04-24:
An easy way to get the full plugin path list is to 

1) install the plugin you want.
2) run 'aprsd list-plugins'

The list-plugins command will show you 3 things.  
1) the built in plugins for aprsd
2) the installable plugins found on pypi.org
3) the current installed plugins and their full plugin path.

```
❯ aprsd list-plugins
                                                       🐍 APRSD Built-in Plugins 🐍
┏━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃ Plugin Name       ┃ Info                                                       ┃ Type         ┃ Plugin Path                             ┃
┡━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│ AVWXWeatherPlugin │ AVWX weather of GPS Beacon location                        │ RegexCommand │ aprsd.plugins.weather.AVWXWeatherPlugin │
│ FortunePlugin     │ Give me a fortune                                          │ RegexCommand │ aprsd.plugins.fortune.FortunePlugin     │
│ NotifySeenPlugin  │ Notify me when a CALLSIGN is recently seen on APRS-IS      │ WatchList    │ aprsd.plugins.notify.NotifySeenPlugin   │
│ OWMWeatherPlugin  │ OpenWeatherMap weather of GPS Beacon location              │ RegexCommand │ aprsd.plugins.weather.OWMWeatherPlugin  │
│ PingPlugin        │ reply with a Pong!                                         │ RegexCommand │ aprsd.plugins.ping.PingPlugin           │
│ TimeOWMPlugin     │ Current time of GPS beacon's timezone. Uses OpenWeatherMap │ RegexCommand │ aprsd.plugins.time.TimeOWMPlugin        │
│ TimePlugin        │ What is the current local time.                            │ RegexCommand │ aprsd.plugins.time.TimePlugin           │
│ USMetarPlugin     │ USA only METAR of GPS Beacon location                      │ RegexCommand │ aprsd.plugins.weather.USMetarPlugin     │
│ USWeatherPlugin   │ Provide USA only weather of GPS Beacon location            │ RegexCommand │ aprsd.plugins.weather.USWeatherPlugin   │
│ VersionPlugin     │ What is the APRSD Version                                  │ RegexCommand │ aprsd.plugins.version.VersionPlugin     │
└───────────────────┴────────────────────────────────────────────────────────────┴──────────────┴─────────────────────────────────────────┘

                                                      Pypi.org APRSD Installable Plugin Packages

                                                       Install any of the following plugins with
                                                          'pip install <Plugin Package Name>'
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━┓
┃ Plugin Package Name          ┃ Description                                                          ┃  Version   ┃      Released       ┃ Installed? ┃
┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━┩
│ 📂 aprsd-assistant-plugin    │ APRSd plugin for hosting the APRS Assistant chatbot (aprs-assistant) │   0.0.3    │ 2024-10-20T02:59:39 │     No     │
│ 📂 aprsd-borat-plugin        │ Borat quotes for aprsd plugin                                        │ 0.1.1.dev1 │ 2024-01-19T16:04:38 │    Yes     │
│ 📂 aprsd-email-plugin        │ APRSD Email plugin                                                   │   1.0.0    │ 2025-01-06T20:48:40 │    Yes     │
│ 📂 aprsd-locationdata-plugin │ Fetch location information from a callsign                           │   0.3.0    │ 2024-02-06T17:20:43 │     No     │
│ 📂 aprsd-mqtt-plugin         │ APRSD MQTT Plugin sends APRS packets to mqtt queue                   │   0.2.0    │ 2023-04-17T16:01:50 │     No     │
│ 📂 aprsd-repeat-plugins      │ APRSD Plugins for the REPEAT service                                 │   1.2.0    │ 2023-01-10T17:15:36 │     No     │
│ 📂 aprsd-sentry-plugin       │ Ham radio APRSD plugin that does....                                 │   0.1.2    │ 2022-12-02T19:07:33 │     No     │
│ 📂 aprsd-slack-plugin        │ Amateur radio APRS daemon which listens for messages and responds    │   1.2.0    │ 2023-01-10T19:21:33 │     No     │
│ 📂 aprsd-stock-plugin        │ Ham Radio APRSD Plugin for fetching stock quotes                     │   0.1.3    │ 2022-12-02T18:56:19 │    Yes     │
│ 📂 aprsd-telegram-plugin     │ Ham Radio APRS APRSD plugin for Telegram IM service                  │   0.1.3    │ 2022-12-02T19:07:15 │     No     │
│ 📂 aprsd-timeopencage-plugin │ APRSD plugin for fetching time based on GPS location                 │   0.2.0    │ 2023-01-10T17:07:11 │     No     │
│ 📂 aprsd-twitter-plugin      │ Python APRSD plugin to send tweets                                   │   0.5.0    │ 2023-01-10T16:51:47 │     No     │
│ 📂 aprsd-weewx-plugin        │ HAM Radio APRSD that reports weather from a weewx weather station.   │   0.3.2    │ 2023-04-20T20:16:19 │     No     │
│ 📂 aprsd-wxnow-plugin        │ APRSD Plugin for getting the closest wx reports to last beacon       │   0.2.0    │ 2023-10-08T01:27:29 │     No     │
└──────────────────────────────┴──────────────────────────────────────────────────────────────────────┴────────────┴─────────────────────┴────────────┘

                                        🐍 APRSD Installed 3rd party Plugins 🐍
[... cut ...]
┡━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│ aprsd-borat-plugin │ BoratPlugin     │        0.1.1        │ RegexCommand │ aprsd_borat_plugin.quotes.BoratPlugin    │
│ aprsd-email-plugin │ EmailPlugin     │ 4.1.3.dev7+g8f471c2 │ RegexCommand │ aprsd_email_plugin.conf.opts.EmailPlugin │
│ aprsd-stock-plugin │ YahooStockQuote │        0.1.3        │ RegexCommand │ aprsd_stock_plugin.stock.YahooStockQuote │
└────────────────────┴─────────────────┴─────────────────────┴──────────────┴──────────────────────────────────────────┘
```