craigerl/aprsd #192: No tx messages

usefulnoise, 2025-04-20:
Hi there,
I'm tying to setup a docker container with arpd but I've some difficulties with reply messages.

This is the file and folder structure:
```
├── aprsd
│   ├── data
│   │   ├── aprsd.conf
│   │   └── save
│   │       └── statsstore.p
│   └── docker-compose.yml
``` 

This is my `docker-compose.yml`
```dockerfile
services:
  aprsd:
    image: hemna6969/aprsd:master
    container_name: aprsd
    ports:
      - 2020:8001
    volumes:
      - ./data:/config
      - ./data/save:/root/.config/aprsd
    restart: unless-stopped
    environment:
      - TZ=Europe/Rome
      - LOG_LEVEL=DEBUG
      - APRSD_PLUGINS=aprsd-admin-extension,aprsd.plugins.fortune.FortunePlugin,aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.version.VersionPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd-assistant-plugin
``` 
This is my `aprsd.conf`
```dockerfile
[DEFAULT]

#
# From aprsd.conf
#

# Callsign to use for messages sent by APRSD (string value)
callsign = CALLSIGN-6

# Enable saving of watch list, packet tracker between restarts.
# (boolean value)
enable_save = true

# Save location for packet tracking files. (string value)
save_location = /root/.config/aprsd/

# Enable code tracing (boolean value)
trace_enabled = true

# Units for display, imperial or metric (string value)
units = metric

# The wait period in seconds per Ack packet being sent.1 means 1 ack
# packet per second allowed.2 means 1 pack packet every 2 seconds
# allowed (integer value)
ack_rate_limit_period = 1

# Wait period in seconds per non AckPacket being sent.2 means 1 packet
# every 2 seconds allowed.5 means 1 pack packet every 5 seconds
# allowed (integer value)
msg_rate_limit_period = 2

# The number of seconds before a packet is not considered a duplicate.
# (integer value)
packet_dupe_timeout = 300

# Enable sending of a GPS Beacon packet to locate this service.
# Requires latitude and longitude to be set. (boolean value)
[... 627 lines cut ...]
aprsd  | RX________(MessagePacket)
aprsd  |   Msg #   : 21
aprsd  |   From    : CALLSIGN-5
aprsd  |   To      : CALLSIGN-6
aprsd  |   Path    : TCPIP*=>qAC=>T2SPAIN
aprsd  |   VIA     : T2SPAIN
aprsd  |   Info    : Fortune
aprsd  |   Raw     : CALLSIGN-5>APZ100::CALLSIGN-6 :Fortune{21
aprsd  | RX________(MessagePacket) | aprsd.packets.log:log_multiline:84
aprsd  | 2025-04-20 18:39:55.028 | ProcessPKT         | DEBUG    | MessagePacket(_type='MessagePacket', from_call='CALLSIGN-5', to_call='CALLSIGN-6', addresse='CALLSIGN-6', format='message', msgNo='21', ackMsgNo=None, packet_type='message', timestamp=1745167195.0, raw='CALLSIGN-5>APZ100::CALLSIGN-6 :Fortune{21', payload=':CALLSIGN-6 :Fortune{21', path=['TCPIP*', 'qAC', 'T2SPAIN'], via='T2SPAIN', message_text='Fortune') | aprsd.packets.log:log_multiline:85
aprsd  | 2025-04-20 18:39:55.031 | ProcessPKT         | DEBUG    | ProcessPKT-LOOP 67 | aprsd.threads.rx:process_packet:210
aprsd  | 2025-04-20 18:39:55.034 | ProcessPKT         | DEBUG    | Packet processing complete for pkt 'CALLSIGN-5:CALLSIGN-6:21' | aprsd.threads.rx:process_packet:266
aprsd  | 2025-04-20 18:39:57.738 | StatsStore         | DEBUG    | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:77
aprsd  | 2025-04-20 18:40:07.744 | StatsStore         | DEBUG    | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:77
aprsd  | 2025-04-20 18:40:09.843 | RX_PKT             | DEBUG    | Server: # aprsc 2.1.19-g730c5c0 20 Apr 2025 16:40:09 GMT T2DENMARK 91.210.59.134:14580 | aprsd.client.drivers.aprsis:consumer:249
aprsd  | 2025-04-20 18:40:17.748 | StatsStore         | DEBUG    | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:77
``` 

</p>
</details>

Also, I can not get trough the aprsd-admin interface using port 2020 nor 8001

Can anybody point me in the right direction?
Many thanks in advance!

hemna, 2025-04-21:
The admin interface is now an extension to aprsd here:

https://github.com/hemna/aprsd-admin-extension

hemna, 2025-04-21:
Also, you have your instance misconfigured.

Your logfile says your callsign is

```
aprsd  | 2025-04-20 18:38:49.671 | MainThread         | DEBUG    | callsign                       = IURIJI-6 | oslo_config.cfg:log_opt_values:2817
```

hemna, 2025-04-21:
Also, are you sure you want to login to the APRS network as 'CALLSIGN-6' ???

usefulnoise, 2025-04-22:
Hi @hemna,
first of all thanks for your time and your patience!

> Also, are you sure you want to login to the APRS network as 'CALLSIGN-6' ???

No, it was a simple placeholder for my callsign 😊

> Also, you have your instance misconfigured.

You are absolute right, and now I'm feeling so dumb for not noticing 😓 (I triple checked my conf file before opening this issue).
Now I can get responses from my instance (thanks again!) 🚀👍 

I edited my docker-compose.yml following your example and it's working with tag 4.0.1 (I'm on amd64 and this tag seems to be the latest one to support my architecture).

Once again, thanks for your help and your support!
Have a lovely day 😉

closed 2025-04-22 by usefulnoise