craigerl/aprsd #194: AVWXWeather

usefulnoise, 2025-04-23:
Hi there,
similar to https://github.com/craigerl/aprsd/issues/127 even if I've configured AVWXWeather, I am still getting a response that the plugin is not enabled.

I'm using the avwx.rest service (hosted) but all I get is:
```shell
aprsd        | 2025-04-23 10:27:05.190 | MainThread         | WARNING  | Plugin AVWXWeatherPlugin is disabled | aprsd.plugin:_load_plugin:460
``` 

<details><summary>docker-compose.yml</summary>

```dockerfile
services:
  aprsd:
    image: hemna6969/aprsd:master
    container_name: aprsd
    volumes:
      - ./data:/config
      - ./data:/root/.config/aprsd
    restart: unless-stopped
    environment:
      - TZ=Europe/Rome
      - LOG_LEVEL=DEBUG
      - APRSD_PLUGINS=aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd.plugins.weather.AVWXWeatherPlugin
  aprsd-admin:
    image: hemna6969/aprsd:4.0.1
    container_name: aprsd-admin
    volumes:
        - ./data:/config
        - ./data:/root/.config/aprsd
    restart: unless-stopped
    ports:
      - 8003:8001
    entrypoint: /app/admin.sh
    environment:
      - TZ=Europe/Rome
      - LOG_LEVEL=DEBUG
      - APRSD_EXTENSIONS=git+https://github.com/hemna/aprsd-admin-extension.git
networks: {}
``` 

</details> 

<details><summary>aprsd.conf</summary>

```
[DEFAULT]

#...

# Comma separated list of enabled plugins for APRSD.To enable
# installed external plugins add them here.The full python path to the
# class name must be used (list value)
enabled_plugins = aprsd.plugins.fortune.FortunePlugin,aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.version.VersionPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd.plugins.weather.AVWXWeatherPlugin

#...

[avwx_plugin]

#
# From aprsd.conf
#

# avwx-api is an opensource project that has a hosted service here:
# https://avwx.rest/You can launch your own avwx-api in a containerby
# cloning the githug repo here:https://github.com/avwx-rest/AVWX-API
# (string value)
apiKey = MYAPIKEYA6u3jRANDOMtextyFuytbWO	

# The base url for the avwx API.  If you are hosting your ownHere is
# where you change the url to point to yours. (string value)
base_url = https://avwx.rest

``` 

</details> 

Has anyone any suggestion on how to enable the plugin using the hosted version of avwx.rest?
Many thanks in advance!

hemna, 2025-04-23:
Do you have any debug logs available?  At initial startup it should explain why it's disabled.

usefulnoise, 2025-04-23:
Here you have 😄 
<details><summary>shell</summary>

```shell
user@localhost:~/docker/aprsd$ docker compose up --force-recreate
[+] Running 3/3
 ✔ Network aprsd_default   Created                                                                                                                      0.1s 
 ✔ Container aprsd-server  Created                                                                                                                      0.1s 
 ✔ Container aprsd-admin   Created                                                                                                                      0.1s 
Attaching to aprsd-admin, aprsd-server
aprsd-admin   | + source /app/.venv/bin/activate
aprsd-admin   | ++ '[' -n x ']'
aprsd-admin   | ++ SCRIPT_PATH=/app/.venv/bin/activate
aprsd-admin   | ++ '[' /app/.venv/bin/activate = /app/admin.sh ']'
aprsd-admin   | ++ deactivate nondestructive
aprsd-admin   | ++ unset -f pydoc
aprsd-admin   | ++ '[' -z '' ']'
aprsd-admin   | ++ '[' -z '' ']'
aprsd-admin   | ++ hash -r
aprsd-admin   | ++ '[' -z '' ']'
aprsd-admin   | ++ unset VIRTUAL_ENV
aprsd-admin   | ++ unset VIRTUAL_ENV_PROMPT
aprsd-admin   | ++ '[' '!' nondestructive = nondestructive ']'
aprsd-admin   | ++ VIRTUAL_ENV=/app/.venv
aprsd-admin   | ++ '[' linux-musl = cygwin ']'
aprsd-admin   | ++ '[' linux-musl = msys ']'
aprsd-admin   | ++ export VIRTUAL_ENV
aprsd-admin   | ++ _OLD_VIRTUAL_PATH=/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
aprsd-admin   | ++ PATH=/app/.venv/bin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
aprsd-admin   | ++ export PATH
aprsd-admin   | ++ '[' xapp '!=' x ']'
aprsd-server  | + source /app/.venv/bin/activate
aprsd-admin   | ++ VIRTUAL_ENV_PROMPT='(app) '
aprsd-server  | ++ '[' -n x ']'
aprsd-admin   | ++ export VIRTUAL_ENV_PROMPT
aprsd-server  | ++ SCRIPT_PATH=/app/.venv/bin/activate
aprsd-admin   | ++ '[' -z '' ']'
aprsd-server  | ++ '[' /app/.venv/bin/activate = /app/setup.sh ']'
aprsd-admin   | ++ '[' -z '' ']'
aprsd-server  | ++ deactivate nondestructive
aprsd-admin   | ++ _OLD_VIRTUAL_PS1=
aprsd-server  | ++ unset -f pydoc
aprsd-admin   | ++ PS1='(app) '
aprsd-server  | ++ '[' -z '' ']'
aprsd-admin   | ++ export PS1
aprsd-server  | ++ '[' -z '' ']'
aprsd-admin   | ++ alias pydoc
aprsd-server  | ++ hash -r
aprsd-admin   | ++ true
aprsd-server  | ++ '[' -z '' ']'
aprsd-admin   | ++ hash -r
aprsd-server  | ++ unset VIRTUAL_ENV
aprsd-admin   | + '[' '!' -z '' ']'
aprsd-server  | ++ unset VIRTUAL_ENV_PROMPT
aprsd-admin   | + '[' '!' -z git+https://github.com/hemna/aprsd-admin-extension.git ']'
aprsd-server  | ++ '[' '!' nondestructive = nondestructive ']'
aprsd-admin   | + OLDIFS=' 	
aprsd-server  | ++ VIRTUAL_ENV=/app/.venv
aprsd-admin   | '
aprsd-server  | ++ '[' linux-musl = cygwin ']'
aprsd-admin   | + IFS=,
aprsd-server  | ++ '[' linux-musl = msys ']'
aprsd-admin   | + echo 'Installing APRSD extensions from pypi '\''git+https://github.com/hemna/aprsd-admin-extension.git'\'''
aprsd-server  | ++ export VIRTUAL_ENV
aprsd-admin   | Installing APRSD extensions from pypi 'git+https://github.com/hemna/aprsd-admin-extension.git'
aprsd-server  | ++ _OLD_VIRTUAL_PATH=/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
aprsd-admin   | + for extension in ${APRSD_EXTENSIONS}
aprsd-server  | ++ PATH=/app/.venv/bin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
aprsd-admin   | + IFS=' 	
aprsd-server  | ++ export PATH
[... 244 lines cut ...]
aprsd-server  | 2025-04-23 15:52:22.671 | MainThread         | DEBUG    | PacketList::No save file found. | aprsd.utils.objectstore:load:113
aprsd-server  | 2025-04-23 15:52:22.672 | MainThread         | DEBUG    | SeenList::No save file found. | aprsd.utils.objectstore:load:113
aprsd-server  | 2025-04-23 15:52:22.672 | MainThread         | DEBUG    | PacketTrack::No save file found. | aprsd.utils.objectstore:load:113
aprsd-server  | 2025-04-23 15:52:22.672 | MainThread         | DEBUG    | WatchList::No save file found. | aprsd.utils.objectstore:load:113
aprsd-server  | 2025-04-23 15:52:22.674 | MainThread         | INFO     | Beacon Enabled.  Starting Beacon thread. | aprsd.cmds.server:server:126
aprsd-server  | 2025-04-23 15:52:22.675 | MainThread         | INFO     | Beacon thread is running and will send beacons every 1800 seconds. | aprsd.threads.tx:__init__:252
aprsd-server  | 2025-04-23 15:52:22.677 | MainThread         | INFO     | Registry Enabled.  Starting Registry thread. | aprsd.cmds.server:server:130
aprsd-server  | 2025-04-23 15:52:22.678 | MainThread         | INFO     | APRS Registry thread is running and will send info every 3600 seconds to https://aprs.hemna.com/api/v1/registry. | aprsd.threads.registry:__init__:30
aprsd-server  | 2025-04-23 15:52:22.680 | KeepAlive          | DEBUG    | Starting | aprsd.threads.aprsd:run:63
aprsd-server  | 2025-04-23 15:52:22.681 | StatsStore         | DEBUG    | Starting | aprsd.threads.aprsd:run:63
aprsd-server  | 2025-04-23 15:52:22.683 | RX_PKT             | DEBUG    | Starting | aprsd.threads.aprsd:run:63
aprsd-server  | 2025-04-23 15:52:22.685 | ProcessPKT         | DEBUG    | Starting | aprsd.threads.aprsd:run:63
aprsd-server  | 2025-04-23 15:52:22.686 | BeaconSendThread   | DEBUG    | Starting | aprsd.threads.aprsd:run:63
aprsd-server  | 2025-04-23 15:52:22.688 | APRSRegistryThread | DEBUG    | Starting | aprsd.threads.aprsd:run:63
[... cut ...]
aprsd-server  | 2025-04-23 15:52:40.692 | StatsStore         | DEBUG    | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:77
aprsd-server  | 2025-04-23 15:52:41.951 | RX_PKT             | DEBUG    | Server: # aprsc 2.1.19-g730c5c0 23 Apr 2025 13:52:41 GMT T2SPAIN 84.232.6.113:14580 | aprsd.client.drivers.aprsis:consumer:249
aprsd-server  | 2025-04-23 15:52:50.697 | StatsStore         | DEBUG    | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:77
``` 

</details> 

Thanks!

hemna, 2025-04-23:
ok I think I have found the issue.  I have it fixed in master

usefulnoise, 2025-04-23:
Terrific! Thanks a lot :)

usefulnoise, 2025-04-24:
Hi @hemna,
I updated the docker image (master) but I still get:
`aprsd-server  | 2025-04-24 17:01:54.694 | MainThread         | WARNING  | Plugin AVWXWeatherPlugin is disabled | aprsd.plugin:_load_plugin:460
`

<details><summary>Full log</summary>

```shell
usefulnoise@quotamare:~/docker/aprsd$ docker compose up --force-recreate
[+] Running 3/3
 ✔ Network aprsd_default   Created                                                                                                                      0.1s 
 ✔ Container aprsd-server  Created                                                                                                                      0.1s 
 ✔ Container aprsd-admin   Created                                                                                                                      0.1s 
Attaching to aprsd-admin, aprsd-server
aprsd-server  | + source /app/.venv/bin/activate
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ '[' -n x ']'
aprsd-server  | ++ SCRIPT_PATH=/app/.venv/bin/activate
aprsd-server  | ++ '[' /app/.venv/bin/activate = /app/setup.sh ']'
aprsd-server  | ++ deactivate nondestructive
aprsd-server  | ++ unset -f pydoc
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ hash -r
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ unset VIRTUAL_ENV
aprsd-server  | ++ unset VIRTUAL_ENV_PROMPT
aprsd-server  | ++ '[' '!' nondestructive = nondestructive ']'
aprsd-server  | ++ VIRTUAL_ENV=/app/.venv
aprsd-server  | ++ '[' linux-musl = cygwin ']'
aprsd-server  | ++ '[' linux-musl = msys ']'
aprsd-server  | ++ export VIRTUAL_ENV
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ unset SCRIPT_PATH
aprsd-server  | ++ _OLD_VIRTUAL_PATH=/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
aprsd-server  | ++ PATH=/app/.venv/bin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
aprsd-server  | ++ export PATH
aprsd-server  | ++ '[' xapp '!=' x ']'
aprsd-server  | ++ VIRTUAL_ENV_PROMPT='(app) '
aprsd-server  | ++ export VIRTUAL_ENV_PROMPT
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ _OLD_VIRTUAL_PS1=
aprsd-server  | ++ PS1='(app) '
aprsd-server  | ++ export PS1
aprsd-server  | ++ alias pydoc
aprsd-server  | ++ true
aprsd-server  | ++ hash -r
aprsd-server  | + COMMAND=server
aprsd-server  | + '[' '!' -z x ']'
aprsd-server  | + COMMAND=server
aprsd-server  | + '[' '!' -z aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd.plugins.weather.AVWXWeatherPlugin,git+https://github.com/hemna/aprsd-email-plugin,git+https://github.com/hemna/aprsd-locationdata-plugin ']'
aprsd-server  | + OLDIFS=' 	
aprsd-server  | '
aprsd-server  | + IFS=,
aprsd-server  | + echo 'Installing pypi plugins '\''aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd.plugins.weather.AVWXWeatherPlugin,git+https://github.com/hemna/aprsd-email-plugin,git+https://github.com/hemna/aprsd-locationdata-plugin'\'''
aprsd-server  | + for plugin in ${APRSD_PLUGINS}
aprsd-server  | + IFS=' 	
aprsd-server  | '
aprsd-server  | + echo 'Installing '\''aprsd.plugins.ping.PingPlugin'\'''
aprsd-server  | + uv pip install aprsd.plugins.ping.PingPlugin
aprsd-server  | Installing pypi plugins 'aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd.plugins.weather.AVWXWeatherPlugin,git+https://github.com/hemna/aprsd-email-plugin,git+https://github.com/hemna/aprsd-locationdata-plugin'
aprsd-server  | Installing 'aprsd.plugins.ping.PingPlugin'
aprsd-admin   | + source /app/.venv/bin/activate
aprsd-admin   | ++ '[' -z '' ']'
aprsd-admin   | ++ '[' -n x ']'
aprsd-admin   | ++ SCRIPT_PATH=/app/.venv/bin/activate
aprsd-admin   | ++ '[' /app/.venv/bin/activate = /app/admin.sh ']'
aprsd-server  | Using Python 3.11.12 environment at: /app/.venv
aprsd-admin   | ++ deactivate nondestructive
[... 342 lines cut ...]
aprsd-server  | 2025-04-24 17:03:52.782 | KeepAlive          | INFO     | Client keepalive 16 seconds ago | aprsd.client.aprsis:keepalive_log:62
aprsd-server  | 2025-04-24 17:03:52.787 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
aprsd-server  | 2025-04-24 17:03:55.817 | RX_PKT             | DEBUG    | Server: # aprsc 2.1.19-g730c5c0 24 Apr 2025 15:03:54 GMT T2UK 192.168.5.41:14580 | aprsd.client.drivers.aprsis:consumer:197
aprsd-server  | 2025-04-24 17:04:02.792 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
aprsd-server  | 2025-04-24 17:04:12.796 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
aprsd-server  | 2025-04-24 17:04:15.832 | RX_PKT             | DEBUG    | Server: # aprsc 2.1.19-g730c5c0 24 Apr 2025 15:04:15 GMT T2UK 192.168.5.41:14580 | aprsd.client.drivers.aprsis:consumer:197
aprsd-server  | 2025-04-24 17:04:22.801 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
[... cut ...]
aprsd-server  | 2025-04-24 17:04:55.861 | RX_PKT             | DEBUG    | Server: # aprsc 2.1.19-g730c5c0 24 Apr 2025 15:04:55 GMT T2UK 192.168.5.41:14580 | aprsd.client.drivers.aprsis:consumer:197
aprsd-server  | 2025-04-24 17:05:02.821 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
aprsd-server  | 2025-04-24 17:05:12.825 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
``` 

</details>

hemna, 2025-04-24:
I think I see the issue.  the aprsd-email-plugin is installing an older version of aprsd 4.0.2, which doesn't have the fix.

hemna, 2025-04-24:
I have updated the requirements.txt for the email plugin.  Just bounce your container.  docker compose down && docker compose up -d