craigerl/aprsd #195: How to activate non built-in plugins?

usefulnoise, 2025-04-23:
Hello,
very dumb question here 🤓 

I'm trying to add some non non built-in plugins to my server.
In my docker-compose.yaml I've add the ones I'm interest in:

```dockerfile
services:
  aprsd:
    image: hemna6969/aprsd:master
    container_name: aprsd-server
    volumes:
      - ./data:/config
      - ./data/:/root/.config/aprsd
    restart: unless-stopped
    environment:
      - TZ=Europe/Rome
      - LOG_LEVEL=DEBUG
      - APRSD_PLUGINS=aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd.plugins.weather.AVWXWeatherPlugin,git+https://github.com/hemna/aprsd-locationdata-plugin,git+https://github.com/hemna/aprsd-email-plugin,git+https://github.com/hemna/aprsd-location-plugin,git+https://github.com/hemna/aprsd-locationdata-plugin,git+https://github.com/hemna/aprsd-digipi-plugin,git+https://github.com/hemna/aprsd-telegram-plugin,git+https://github.com/hemna/aprsd-repeat-plugins
``` 
How do I activate them in the `aprsd.config` file?

<details><summary>Logs here</summary>

```shell
user@localhost:~/docker/aprsd$ docker compose up --force-recreate
[+] Running 3/3
 ✔ Network aprsd_default   Created                                                                                                                      0.1s 
 ✔ Container aprsd-server  Created                                                                                                                      0.1s 
 ✔ Container aprsd-admin   Created                                                                                                                      0.1s 
Attaching to aprsd-admin, aprsd-server
aprsd-server  | + source /app/.venv/bin/activate
aprsd-server  | ++ '[' -n x ']'
aprsd-server  | ++ SCRIPT_PATH=/app/.venv/bin/activate
aprsd-server  | ++ '[' /app/.venv/bin/activate = /app/setup.sh ']'
aprsd-server  | ++ deactivate nondestructive
aprsd-server  | ++ unset -f pydoc
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ hash -r
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ unset VIRTUAL_ENV
aprsd-server  | ++ unset VIRTUAL_ENV_PROMPT
aprsd-server  | ++ '[' '!' nondestructive = nondestructive ']'
aprsd-server  | ++ VIRTUAL_ENV=/app/.venv
aprsd-server  | ++ '[' linux-musl = cygwin ']'
aprsd-server  | ++ '[' linux-musl = msys ']'
aprsd-server  | ++ export VIRTUAL_ENV
aprsd-server  | ++ _OLD_VIRTUAL_PATH=/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
aprsd-server  | ++ PATH=/app/.venv/bin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
aprsd-server  | ++ export PATH
aprsd-server  | ++ '[' xapp '!=' x ']'
aprsd-server  | ++ VIRTUAL_ENV_PROMPT='(app) '
aprsd-server  | ++ export VIRTUAL_ENV_PROMPT
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ '[' -z '' ']'
aprsd-server  | ++ _OLD_VIRTUAL_PS1=
aprsd-server  | ++ PS1='(app) '
aprsd-server  | ++ export PS1
aprsd-server  | ++ alias pydoc
aprsd-server  | ++ true
aprsd-server  | ++ hash -r
aprsd-server  | + COMMAND=server
aprsd-server  | + '[' '!' -z x ']'
aprsd-server  | + COMMAND=server
aprsd-server  | + '[' '!' -z aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd.plugins.weather.AVWXWeatherPlugin,git+https://github.com/hemna/aprsd-locationdata-plugin,git+https://github.com/hemna/aprsd-email-plugin,git+https://github.com/hemna/aprsd-location-plugin,git+https://github.com/hemna/aprsd-locationdata-plugin,git+https://github.com/hemna/aprsd-digipi-plugin,git+https://github.com/hemna/aprsd-telegram-plugin,git+https://github.com/hemna/aprsd-repeat-plugins ']'
aprsd-server  | + OLDIFS=' 	
aprsd-server  | '
aprsd-server  | + IFS=,
aprsd-server  | + echo 'Installing pypi plugins '\''aprsd.plugins.ping.PingPlugin,aprsd.plugins.time.TimePlugin,aprsd.plugins.weather.OWMWeatherPlugin,aprsd.plugins.notify.NotifySeenPlugin,aprsd.plugins.weather.AVWXWeatherPlugin,git+https://github.com/hemna/aprsd-locationdata-plugin,git+https://github.com/hemna/aprsd-email-plugin,git+https://github.com/hemna/aprsd-location-plugin,git+https://github.com/hemna/aprsd-locationdata-plugin,git+https://github.com/hemna/aprsd-digipi-plugin,git+https://github.com/hemna/aprsd-telegram-plugin,git+https://github.com/hemna/aprsd-repeat-plugins'\'''
[... 394 lines cut ...]
aprsd-server  | 2025-04-23 16:43:10.486 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
aprsd-server  | 2025-04-23 16:43:12.356 | RX_PKT             | DEBUG    | Server: # aprsc 2.1.19-g730c5c0 23 Apr 2025 14:43:12 GMT T2GB 192.146.137.137:14580 | aprsd.client.drivers.aprsis:consumer:197
aprsd-server  | 2025-04-23 16:43:20.491 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
aprsd-server  | 2025-04-23 16:43:30.476 | KeepAlive          | INFO     | IU3IJI-6 - Uptime 0:00:50.005971 RX:0 TX:0 Tracker:0 Msgs TX:0 RX:0 Last:N/A - RAM Current:32KB Peak:54KB Threads:6 LoggingQueue:0 | aprsd.threads.keepalive:loop:76
aprsd-server  | 2025-04-23 16:43:30.478 | KeepAlive          | INFO     | KeepAlive       Alive? True  0:00:00.000639       | aprsd.threads.keepalive:loop:90
aprsd-server  | 2025-04-23 16:43:30.480 | KeepAlive          | INFO     | StatsStore      Alive? True  0:00:00.981937       | aprsd.threads.keepalive:loop:90
aprsd-server  | 2025-04-23 16:43:30.482 | KeepAlive          | INFO     | RX_PKT          Alive? True  0:00:00.106979       | aprsd.threads.keepalive:loop:90
aprsd-server  | 2025-04-23 16:43:30.484 | KeepAlive          | INFO     | ProcessPKT      Alive? True  0:00:00.991111       | aprsd.threads.keepalive:loop:90
aprsd-server  | 2025-04-23 16:43:30.486 | KeepAlive          | INFO     | BeaconSendThread Alive? True  0:00:00.991039       | aprsd.threads.keepalive:loop:90
aprsd-server  | 2025-04-23 16:43:30.488 | KeepAlive          | INFO     | APRSRegistryThread Alive? True  0:00:00.987576       | aprsd.threads.keepalive:loop:90
aprsd-server  | 2025-04-23 16:43:30.495 | StatsStore         | INFO     | StatsStore::Saving 8 entries to disk at /root/.config/aprsd//statsstore.p | aprsd.utils.objectstore:save:78
aprsd-server  | 2025-04-23 16:43:30.497 | KeepAlive          | INFO     | Client keepalive 18 seconds ago | aprsd.client.aprsis:keepalive_log:62
aprsd-server  | 2025-04-23 16:43:32.370 | RX_PKT             | DEBUG    | Server: # aprsc 2.1.19-g730c5c0 23 Apr 2025 14:43:32 GMT T2GB 192.146.137.137:14580 | aprsd.client.drivers.aprsis:consumer:197
a
``` 

</details> 

Also, can you share the configuration parts for the Telegram and the email plugins?
I didn't found anything on the sample-config file 😕 

I tried checking the repositories of the various plugins to see if there was any documentation, but I couldn’t find anything.
If you’ve already written something, sorry about that, could you please point me to it?

Thanks once again for your time and your patience!

usefulnoise, 2025-04-23:
Hey @hemna 

> I guess I should/could do a better job with documentation, but I'm only 1 person, so there is only so much I can do.

Not at all, you're doing a terrific job maintaining this application and supporting everyone needs help! 🎉 

I was able to install in a dev env all the plugins and generate the `aprsd sample-config` file, thanks again for your input 😉 

Unfortunately, once I've installed all the plugins, if I run `aprsd list-plugins` in order to get the Plugin Path I get:
```shell
(aprsd) user@aprs:~ $ aprsd list-plugins
                                                       🐍 APRSD Built-in Plugins 🐍                                                        
┏━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃ Plugin Name       ┃ Info                                                       ┃ Type         ┃ Plugin Path                             ┃
┡━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│ AVWXWeatherPlugin │ AVWX weather of GPS Beacon location                        │ RegexCommand │ aprsd.plugins.weather.AVWXWeatherPlugin │
│ FortunePlugin     │ Give me a fortune                                          │ RegexCommand │ aprsd.plugins.fortune.FortunePlugin     │
│ NotifySeenPlugin  │ Notify me when a CALLSIGN is recently seen on APRS-IS      │ WatchList    │ aprsd.plugins.notify.NotifySeenPlugin   │
│ OWMWeatherPlugin  │ OpenWeatherMap weather of GPS Beacon location              │ RegexCommand │ aprsd.plugins.weather.OWMWeatherPlugin  │
│ PingPlugin        │ reply with a Pong!                                         │ RegexCommand │ aprsd.plugins.ping.PingPlugin           │
│ TimeOWMPlugin     │ Current time of GPS beacon's timezone. Uses OpenWeatherMap │ RegexCommand │ aprsd.plugins.time.TimeOWMPlugin        │
│ TimePlugin        │ What is the current local time.                            │ RegexCommand │ aprsd.plugins.time.TimePlugin           │
│ USMetarPlugin     │ USA only METAR of GPS Beacon location                      │ RegexCommand │ aprsd.plugins.weather.USMetarPlugin     │
│ USWeatherPlugin   │ Provide USA only weather of GPS Beacon location            │ RegexCommand │ aprsd.plugins.weather.USWeatherPlugin   │
│ VersionPlugin     │ What is the APRSD Version                                  │ RegexCommand │ aprsd.plugins.version.VersionPlugin     │
└───────────────────┴────────────────────────────────────────────────────────────┴──────────────┴─────────────────────────────────────────┘
Traceback (most recent call last):
  File "/home/user/venv/aprsd/bin/aprsd", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/aprsd/main.py", line 72, in main
    cli(auto_envvar_prefix="APRSD")
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/click/core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/aprsd/cli_helper.py", line 156, in new_func
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/aprsd/cmds/list_plugins.py", line 305, in list_plugins
    installed_plugins = get_installed_plugins()
                        ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/aprsd/cmds/list_plugins.py", line 109, in get_installed_plugins
    plugins, extensions = _get_installed_aprsd_items()
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/aprsd/cmds/list_plugins.py", line 97, in _get_installed_aprsd_items
    pkg_info = get_module_info(
               ^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/aprsd/cmds/list_plugins.py", line 72, in get_module_info
    module = smuggle(f'{path}/{name}')
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/thesmuggler.py", line 50, in smuggle
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/user/venv/aprsd/lib/python3.11/site-packages/aprsd_email_plugin/aprsd_email_plugin.py", line 14, in <module>
    from aprsd.threads import keepalive_collector, tx
ImportError: cannot import name 'keepalive_collector' from 'aprsd.threads' (/home/user/venv/aprsd/lib/python3.11/site-packages/aprsd/threads/__init__.py)
``` 
Can you just share with me the Plugin Path?

Thanks again!

hemna, 2025-04-23:
hrmm, are you running aprsd from master?

hemna, 2025-04-23:
The other issue I just noticed is for non built in plugins the list-plugins command doesn't provide the Plugin Path, even when installed.

hemna, 2025-04-23:
```
❯ aprsd list-plugins
                                                       🐍 APRSD Built-in Plugins 🐍
┏━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃ Plugin Name       ┃ Info                                                       ┃ Type         ┃ Plugin Path                             ┃
┡━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│ AVWXWeatherPlugin │ AVWX weather of GPS Beacon location                        │ RegexCommand │ aprsd.plugins.weather.AVWXWeatherPlugin │
│ FortunePlugin     │ Give me a fortune                                          │ RegexCommand │ aprsd.plugins.fortune.FortunePlugin     │
│ NotifySeenPlugin  │ Notify me when a CALLSIGN is recently seen on APRS-IS      │ WatchList    │ aprsd.plugins.notify.NotifySeenPlugin   │
│ OWMWeatherPlugin  │ OpenWeatherMap weather of GPS Beacon location              │ RegexCommand │ aprsd.plugins.weather.OWMWeatherPlugin  │
│ PingPlugin        │ reply with a Pong!                                         │ RegexCommand │ aprsd.plugins.ping.PingPlugin           │
│ TimeOWMPlugin     │ Current time of GPS beacon's timezone. Uses OpenWeatherMap │ RegexCommand │ aprsd.plugins.time.TimeOWMPlugin        │
│ TimePlugin        │ What is the current local time.                            │ RegexCommand │ aprsd.plugins.time.TimePlugin           │
│ USMetarPlugin     │ USA only METAR of GPS Beacon location                      │ RegexCommand │ aprsd.plugins.weather.USMetarPlugin     │
│ USWeatherPlugin   │ Provide USA only weather of GPS Beacon location            │ RegexCommand │ aprsd.plugins.weather.USWeatherPlugin   │
│ VersionPlugin     │ What is the APRSD Version                                  │ RegexCommand │ aprsd.plugins.version.VersionPlugin     │
└───────────────────┴────────────────────────────────────────────────────────────┴──────────────┴─────────────────────────────────────────┘

                                                      Pypi.org APRSD Installable Plugin Packages

                                                       Install any of the following plugins with
                                                          'pip install <Plugin Package Name>'
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━┓
┃ Plugin Package Name          ┃ Description                                                          ┃  Version   ┃      Released       ┃ Installed? ┃
┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━┩
│ 📂 aprsd-assistant-plugin    │ APRSd plugin for hosting the APRS Assistant chatbot (aprs-assistant) │   0.0.3    │ 2024-10-20T02:59:39 │     No     │
│ 📂 aprsd-borat-plugin        │ Borat quotes for aprsd plugin                                        │ 0.1.1.dev1 │ 2024-01-19T16:04:38 │    Yes     │
│ 📂 aprsd-email-plugin        │ APRSD Email plugin                                                   │   1.0.0    │ 2025-01-06T20:48:40 │    Yes     │
│ 📂 aprsd-locationdata-plugin │ Fetch location information from a callsign                           │   0.3.0    │ 2024-02-06T17:20:43 │     No     │
│ 📂 aprsd-mqtt-plugin         │ APRSD MQTT Plugin sends APRS packets to mqtt queue                   │   0.2.0    │ 2023-04-17T16:01:50 │     No     │
│ 📂 aprsd-repeat-plugins      │ APRSD Plugins for the REPEAT service                                 │   1.2.0    │ 2023-01-10T17:15:36 │     No     │
│ 📂 aprsd-sentry-plugin       │ Ham radio APRSD plugin that does....                                 │   0.1.2    │ 2022-12-02T19:07:33 │     No     │
│ 📂 aprsd-slack-plugin        │ Amateur radio APRS daemon which listens for messages and responds    │   1.2.0    │ 2023-01-10T19:21:33 │     No     │
│ 📂 aprsd-stock-plugin        │ Ham Radio APRSD Plugin for fetching stock quotes                     │   0.1.3    │ 2022-12-02T18:56:19 │     No     │
│ 📂 aprsd-telegram-plugin     │ Ham Radio APRS APRSD plugin for Telegram IM service                  │   0.1.3    │ 2022-12-02T19:07:15 │     No     │
│ 📂 aprsd-timeopencage-plugin │ APRSD plugin for fetching time based on GPS location                 │   0.2.0    │ 2023-01-10T17:07:11 │     No     │
│ 📂 aprsd-twitter-plugin      │ Python APRSD plugin to send tweets                                   │   0.5.0    │ 2023-01-10T16:51:47 │     No     │
│ 📂 aprsd-weewx-plugin        │ HAM Radio APRSD that reports weather from a weewx weather station.   │   0.3.2    │ 2023-04-20T20:16:19 │     No     │
│ 📂 aprsd-wxnow-plugin        │ APRSD Plugin for getting the closest wx reports to last beacon       │   0.2.0    │ 2023-10-08T01:27:29 │     No     │
└──────────────────────────────┴──────────────────────────────────────────────────────────────────────┴────────────┴─────────────────────┴────────────┘

                                      🐍 APRSD Installed 3rd party Plugins 🐍
┏━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃ Package Name       ┃ Plugin Name ┃       Version       ┃ Type         ┃ Plugin Path                              ┃
┡━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│ aprsd-borat-plugin │ BoratPlugin │        0.1.1        │ RegexCommand │ aprsd_borat_plugin.quotes.BoratPlugin    │
│ aprsd-email-plugin │ EmailPlugin │ 4.1.3.dev7+g8f471c2 │ RegexCommand │ aprsd_email_plugin.conf.opts.EmailPlugin │
└────────────────────┴─────────────┴─────────────────────┴──────────────┴──────────────────────────────────────────┘
```

usefulnoise, 2025-04-24:
> hrmm, are you running aprsd from master?
> 
> Did you install the email plugin from pypi or from git?
> 
> If you are running aprsd from github master, then you'll have to install the email plugin from git as well. I'm in the process still of doing some refactoring for the next release. The github aprsd_email_plugin has accounted for the changes already...hopefully.

I used a Raspberry Pi as development system installing everything from pip (both master and 4.0.1).