craigerl/aprsd #196: cant visit admin web page

FengziLeo, 2025-05-12:
I run aprsd in the Docker python:3.11 
install aprsd by "pip install aprsd" and install aprsd-admin-extension
use "aprsd server --config /config/aprsd.conf --loglevel DEBUG" to run aprsd
here is log and conf
[log.txt](https://github.com/user-attachments/files/20153879/log.txt)

[aprsd (1).conf.txt](https://github.com/user-attachments/files/20153882/aprsd.1.conf.txt)

I also set up Docker to map port 8001 to the host, but accessing ip:8001 does not display any content.

hemna, 2025-06-06:
The admin interface is part of the admin extension that has to be run separately.

This is how I run it as a separate container.

https://github.com/craigerl/aprsd/blob/master/docker/bin/admin.sh#L48

and part of my docker-compose.yml

```

  aprsd-wb4bor:
    image: hemna6969/aprsd:master
    container_name: aprsd-wb4bor-11
    depends_on: [avwx-api]
    links: [avwx-api]
    ports:
      - 18863:18863
    volumes:
        - /home/waboring/docker/aprsd-wb4bor-11/config:/config
    restart: unless-stopped
    environment:
        - TZ=America/New_York
        - APRSD_PLUGINS=git+https://github.com/hemna/aprsd-slack-plugin.git,git+https://github.com/hemna/aprsd-telegram-plugin.git,aprsd-stock-plugin,git+https://github.com/hemna/aprsd-w3w-plugin.git,git+https://github.com/hemna/aprsd-borat-plugin.git,git+https://github.com/hemna/aprsd-email-plugin.git,git+https://github.com/hemna/aprsd-location-plugin.git
    healthcheck:
        #test:  aprsd healthcheck --config /config/aprsd.conf --loglevel DEBUG --timeout 5
        test:  /app/healthcheck.sh
        interval: 1m
        timeout: 5s
    labels:
        com.centurylinklabs.watchtower.enable: "false"
        deunhealth.restart.on.unhealthy: "true"

  aprsd-admin:
    image: hemna6969/aprsd:master
    container_name: aprsd-wb4bor-11-admin
    volumes:
        - /home/waboring/docker/aprsd-wb4bor-11/config:/config
    restart: unless-stopped
    ports:
      - 8003:8001
    entrypoint: /app/admin.sh
    labels:
        com.centurylinklabs.watchtower.enable: "false"
        deunhealth.restart.on.unhealthy: "true"
    environment:
      - TZ=America/New_York
      - APRSD_EXTENSIONS=git+https://github.com/hemna/aprsd-admin-extension.git
```

FengziLeo, 2025-06-07:
> The admin interface is part of the admin extension that has to be run separately.
> 
> This is how I run it as a separate container.
> 
> https://github.com/craigerl/aprsd/blob/master/docker/bin/admin.sh#L48
> 
> and part of my docker-compose.yml

Thank you very much! I have successfully run it. However, I have one possibly silly question: how should I send APRS messages to aprsd? I am not sure how to fill in the APRS server address, and whether Docker needs to expose any ports, and if so, which port. I saw you mentioned port 18863, but I tried it and still couldn't receive messages properly, so I would like to ask how you suggest setting it up.

hemna, 2025-06-07:
For what it's worth, you can use aprsd webchat web interface to send messages to your aprsd server instance too.

https://github.com/hemna/aprsd-webchat-extension

hemna, 2025-06-09:
Yes, APRSD server command acts like a bot/autoresponder on the APRS network.  It receives messages to it's callsign, and sends those messages through the list of plugins that are enabled.  If the message contents match what the plugin is looking for, it will create a message to respond to the caller.  

APRSD Has other functions and capabilities as well you might want to check out on the command line.   

aprsd --help

the admin interface itself is an extension (adds new capabilities) to aprsd.  So is webchat.

FengziLeo, 2025-06-10:
Roger! Thanks for your answer~73